black-parrot/black-parrot #504: Makefile.verilator depends on Bash 4

muwyse, 2020-05-20:
The `|&` operator, which is equivalent to `2>&1 |` is used in Makefile.verilator, but is only supported by Bash 4 and later. By default, many systems link the dash shell to /bin/sh, which apparently does not necessarily support the `|&` operator. This only manifests if /bin/sh points to something that doesn't support the operator, for example, dash, which appears to be the default on my system (Ubuntu on WSL / Windows 10).

code: https://github.com/black-parrot/black-parrot/blob/master/bp_common/syn/Makefile.verilator#L50

Question: should this operator be removed and use `2>&1 |` instead, or are there other features from Bash 4 that we use, and thus want to make Bash 4 (or equivalent) an explicit requirement?

Note: it is easy for me to reconfigure my system to point /bin/sh at bash, but this may not be true for all users, especially if they are running on managed machines they do not have full control over.

dpetrisko, 2020-05-21:
Interesting.  My opinion on this is to generally to provide a pathway for everyone, but only promise to maintain what we can.  The usecases I see are:

- Users who don't control their machine (academics, corporate users), who tend to be using Ubuntu-derivatives or CentOS.  We want these to work out-of-box.  Arguably Windows fits in this category, but it's always pretty difficult to maintain.
- Users who control their machines (hobbyists, smaller ASIC shops), who could be using Mac, Windows, random Linux distros.

As long as we're compatible with CentOS 7+ and Ubuntu 16.04+, I think it's reasonable to suggest some directions for common other setups (upgrade to Bash 4) or just provide a Docker image for more unusual use cases.

@taylor-bsg , what are your thoughts on this?

muwyse, 2020-05-21:
I agree. Generally provide path for all, but promise to support what we can.

In this particular case, I think we should make the change and revert it to `2>&1 |` since there doesn't seem to be any obvious benefit why we need the Bash 4 syntax over the old style syntax. Doing so removes a low-cost barrier for enabling use.

dpetrisko, 2020-05-21:
Fine by me

PR #510 (2020-05-21): replace |& operator with 2>&1 | for greater compatibility
closes #504
closed 2020-05-21 by muwyse