black-parrot/black-parrot #512: Waveform no rule to make dirs.sc

chaserwhytey, 2020-05-23:
Problem with getting waveforms, running command
make wave.sc SUITE=bp_tests PROG=hello_world
No rule to make target 'dirs.sc

chaserwhytey, 2020-05-23:
I'm also seeing no rule for making target sim_dump.sc in the current Makefile.verilator

dpetrisko, 2020-05-23:
Thanks for the issue! Will update in the next release.  The fixes to Makefile.verilator are shown below.

sim_dump.sc was added in dev recently: https://github.com/black-parrot/black-parrot/blob/dev/bp_common/syn/Makefile.verilator#L87

```
sim_dump.sc: VV_OPTS += --trace
sim_dump.sc: VV_OPTS += --trace-structs
sim_dump.sc: VV_OPTS += --trace-depth 15
sim_dump.sc: sim.sc
```

```
wave.sc: $(SIM_DIR)/wavesc
$(SIM_DIR)/wavesc:
	cd $(@D); $(GTKWAVE) -f dump.vcd &

# TODO: Make compatible with multi-core
BLOODGRAPH ?= $(BP_COMMON_DIR)/software/py/blood_graph.py --generate --generate-key
blood.sc: $(SIM_DIR)/bloodsc
$(SIM_DIR)/bloodsc:
	cd $(@D); $(PYTHON) $(BLOODGRAPH) --input $(@D)/stall_0.trace
```

PR #515 (2020-05-26): Updating makefiles to play nicer with waveform viewers
fixes #512
closed 2020-05-26 by black-parrot