black-parrot/black-parrot #521: Stressing branch misprediction logic (verification experiment)

kabylkas, 2020-05-27:
Hi :wave:,

I was testing another verification hypothesis. What I am doing is basically generating random legal instruction addresses out of BTB. My assumption is that branch prediction mechanism should not affect functionality of the core and doing this will simply trigger a lot of mispredictions. However, Black Parrot hangs for some tests (watchdog timeout). I was wondering if doing this is okay in Black Parrot?

Thanks!

[image]

dpetrisko, 2020-05-27:
Hi! Yep, that should be legal.  Could you share an address that's causing a hang? It's possible that it's generating an address that goes to unmapped local memory, which would be < 0x8000_0000 and not mapping to cfg, clint or cache MMIO CSRs.  If a request is sent to an unmapped address region, it could cause a hang. 

Another unusual case would be > 0x20_0000_0000, which maps off-chip memory.  This should cause a speculative instruction fault, which will get squashed before causing a trap (since it's speculative).

In the former case, a patch was merged for this earlier today.  https://github.com/black-parrot/black-parrot/pull/519.  If it's a non-local address, we'll have to look deeper!

kabylkas, 2020-05-27:
I am trying to mimic the legal instruction addresses, meaning the addresses are half-word aligned and are in the range 0x8000_0000 to 0x8000_FFFF.

I am trying to isolate the address that causes the hang now, but I have just realized that I might be doing something wrong. Quick question: ports `br_tgt_v_o` and `br_tgt_jmp_o` in `bp_fe_btb` can be randomized as well, can't they?

dpetrisko, 2020-05-27:
I would do word-aligned.  We don't have support for misaligned addresses (hardware or trapping) at the moment.

If br_tgt_v_o is forced high when the entry is invalid, that could cause X-propagation.  Similar to our other handshaking discussion, it's generally only allowed to mask helpful signals, not force them high.

I think br_tgt_jmp_o can be freely randomized (subject to the range and alignment restrictions)

dpetrisko, 2023-07-21:
Closing for staleness

closed 2023-07-21 by dpetrisko