black-parrot/black-parrot #530: Test files are all over the place

billythedummy, 2020-06-01:
In the case of bp_tests, running
```
make -C bp_common/test
```
in the root folder causes `.riscv` binary test files to be outputted to `bp_common/test/src/bp_tests`. 

However, 
```
make sim.sc PROG=$PROG
```
seems to use the $PROG.riscv located in `bp_common/test/mem/bp_tests` to generate the `.mem` files to copy into the simulation. This results in the binary being ran in simulation not being updated by test src file updates. 

And then I think there's some other make command that outputs `.riscv` and `.mem` files to `bp_common/test/mem`?

dpetrisko, 2020-06-01:
Hi, thanks for the issue. We should document this flow more  

All test suites are generated to bp_common/test/mem. All test suites are versioned by submodule and only build when out of date. In order to force a test build use, for instance, 
`make -C bp_common/test bp_tests_manual`

billythedummy, 2020-06-01:
Thank you for the quick reply! I haven't had time to dig through all the Makefiles either but may I also ask why the same `.riscv` binary is being copied across 3 or 4 different locations?

dpetrisko, 2020-06-01:
There are only two. The flow is that make -C <suite> is called to generate .riscv and then they are copied to bp_common/test/mem/<suite> folder to be used in simulation.  They could be mv'd instead of cp'd, but the space savings is meh, and leaving them there means that they won't be rebuilt again if nothing has changed

dpetrisko, 2020-06-12:
"How to add a test" documentary was added with https://github.com/black-parrot/black-parrot/pull/537

closed 2020-06-12 by dpetrisko