black-parrot/black-parrot #534: ptag_width_p used incorrectly in caches and LCE's

muwyse, 2020-06-02:
In bp_common_aviary_defines.vh the page_offest_width_p param is hard-coded to 12-bits (4K pages), and then used to define ptag_width_p = paddr_width - page_offset_width_p. This parameter appears to be used in the icache, dcache, and LCE's, regardless of what their specific organizations are. Thus it is possible to configure the caches differently such that the actual tag size for the cache does not match ptag_width_p.

This works as long as **every** cache in the system obeys this restriction and is organized such that number of set index bits + block offset bits == 12. However, if we wish to use an icache or dcache organization that results in a violation of this constraint, we should expect undefined behavior.

I suspect this ptag_width_p param is intended for use in the TLB's and VM system, where this width represents the PPN width.

dpetrisko, 2020-06-02:
Issue makes sense.  The ptag width is set in the caches such that there is no synonym problem in VIPT caches.  ptag_width is also a sufficient subset of ppn bits.  Since we support 40 < 56 bit physical address, this the width of the structs which the are stored in the TLBs and passed around the system.  So ptag_width is relevant for any cache which is VIPT.

The current I$ and D$ are only compatible with virtual page_offset = 12 bits, physical tag, which is a reasonable constraint until we decide to violate it (probably in the I$, since synonyms are not as big of a problem).  However, I agree accelerator caches, for example, will not need this restriction, especially if they are PIPT.  Since the accelerator cache is currently a D$, this means the dcache probably needs variable tag support.

Maybe the thing to do is generate tag_width_lp in the caches and xCEs based on organization.  And make an assertion for the I$ and D$ that tag_width_lp == ptag_width_p, for now.  Seem reasonable?

taylor-bsg, 2020-06-08:
See also:

https://github.com/black-parrot/black-parrot/issues/394

https://github.com/black-parrot/black-parrot/issues/393

taylor-bsg, 2020-06-08:
https://github.com/black-parrot/black-parrot/issues/392

dpetrisko, 2020-11-04:
fixed in #658

closed 2020-11-04 by dpetrisko