black-parrot/black-parrot #557: Investigate impact of supporting L2 atomics in BedRock

dpetrisko, 2020-06-29:
L2 atomics are useful for highly contended aggregation, such as parallel reduction. We should support this option in BedRock, so that LCEs can request the CCE to perform an atomic operation

Changes required:
[ ] Add atomic message types to LCE/CCE IF
[ ] Add atomic message types of CCE/MEM IF
[ ] Add parameterized atomic operation to L1
[ ] Integrate bsg_cache atomics from this PR [](https://github.com/bespoke-silicon-group/basejump_stl/pull/298)

Other possible features:
[ ] Atomic support with CCE ucode
[ ] Add flag for "no return" atomics (rd == x0) which are non-blocking

muwyse, 2020-07-22:
See this branch for initial support in the FSM CCE: https://github.com/black-parrot/black-parrot/tree/l2_atomics_cce, which builds on branch https://github.com/black-parrot/black-parrot/tree/feature_enable_l2atomics

dpetrisko, 2021-02-25:
Partially fixed with #765

dpetrisko, 2023-07-18:
All are implemented except for L2 LR/SC, so closing for now

closed 2023-07-18 by dpetrisko