black-parrot/black-parrot #562: How to connect Black Parrot tiles to Celerity-like manycore accelerator?

qbcir, 2020-07-01:
Hello,

first, thank you all for sharing such a great RISC-V core with world. While connecting BP cores into multicore tiles array is straightforward, it's not clear which interfaces and IPs should one use to connect BP multicore to celerity-like manycore accelerator. By "celerity-like" I mean use NoC from bsg_manycore and some lightweight cores (SIMT cores for example). Could you help, please?

dpetrisko, 2020-07-01:
Hi! Great to hear you're interested in the project.

Connecting a single BP core to a manycore accelerator can be done in a few ways.  The most straightforward way is to simply attach the BlackParrot I/O port to a "manycore_bridge" which converts between the ```bp_cce_mem_msg_s``` message format and the manycore network interface.  I have a proof of concept that I was working on for a class project here: [bp_to_manycore_link](https://github.com/dpetrisko/bsg_manycore/blob/master/machines/4x4_hammerblade/bp_cce_to_mc.v)

For a BP multicore to bsg_manycore, the simplest way to attach is by connecting an array of these same manycore_bridges in the streaming accelerator complex, or "SAC" We have an example in the repo of using a dot-product accelerator in the SAC: [dot-product integration](https://github.com/black-parrot/black-parrot/blob/master/bp_top/src/v/bp_sacc_tile.v).  This will allow you to connect the manycore_bridge to a BedRock router and ```bp_io_cce```, which has an output of ```bp_cce_mem_msg_s``` that you can transduce to a manycore NoC format, in the same way as the single core case.  Depending on how you want to handle your SoC hierarchy, you could encompass the manycore within the SAC, or connect new manycore ports that hang off the side of the BP chip.

We're extremely excited about this type of integration, so do let us know how it goes for your project! We're also pursuing this, and plan on updating the docs with lessons learned and best practices as we go along

dpetrisko, 2023-07-18:
Closing for staleness; but you can see progress here: https://github.com/black-parrot-hdk/zynq-parrot/tree/80b6db0734fd1093c323bf0003a86775f59dc2f2/cosim/hammerblade-example

closed 2023-07-18 by dpetrisko