black-parrot/black-parrot #578: Accepting responses in uncached-only mode in the microcode CCE

sripathi-muralitharan, 2020-07-19:
There seems to be a bug in operating the CCE in the uncached-only mode.
As of now, in uncached mode the CCE can accept memory responses only when in the `uc_ready` state. This would mean that if we have a stream of requests and the L2's FIFO is full, the uncached FSM in the CCE can get stuck in another state different from `uc_ready` in which case the entire system will be locked up. Therefore, the solution is to have the uncached mode also accept responses irrespective of state as is done in the cached mode. 

To reproduce this bug, comment out the line corresponding to changing the CCE mode in the `nbf.py` script in `bp_common/software/py` and then running any program. This would force the program to be loaded into the DRAM in uncached-only mode where this bug can be observed. I used the `e_bp_single_core_ucode_cce_cfg` config and ran the `hello_world` program. I observed that the uncached-only FSM in the message module is stuck in the `uc_send_data_command` state. 

I did not find this bug with the FSM-based CCE most likely because of the way it has been implemented.

muwyse, 2020-07-20:
Good catch. Two solutions here:
1. do as suggested and move the memory response logic out of the uc_ready state and allow the ucode CCE in uncached only mode to process responses regardless of its current state.
2. add a limit on the number of outstanding memory commands/accesses while in the uncached only mode. This then requires the system designer to ensure enough buffering between the CCE and memory endpoint / within the network to allow the CCE to issue its maximum number of commands.

Since the LCE Command port is only used to send uncached data or uncached access complete messages to the LCE when memory responses return while the CCE is in uncached only mode, I suggest option 1. Since uncached only mode is a mode where coherence is not provided, there are no other potentially conflicting uses of the LCE Command port that would require arbitration logic between automatically processing the memory response and some other FSM state trying to use the port.

It is still possible (and may be something we want to do) to add a maximum outstanding request limit to this approach, too, which is a one line change to the mem_cmd_o logic in uc_ready. This may be desirable to allow system designers to flow control the traffic out of the CCE. A similar change may be desirable in the coherent/microcode side of the CCE, too.

dpetrisko, 2020-07-20:
I'd strongly prefer 1 as well, if it's possible to do without much overhead.

Maximum request counters are also okay, using a credit-based approach

muwyse, 2020-07-20:
@sripathi-muralitharan see the changes on this branch for a fix:

muwyse, 2020-07-20:
See this PR: https://github.com/black-parrot/black-parrot/pull/579

muwyse, 2020-07-22:
Should be fixed by #579. Closing issue.

closed 2020-07-22 by muwyse