black-parrot/black-parrot #581: Track memory network credits in ucode CCE

dpetrisko, 2020-07-21:
https://github.com/black-parrot/black-parrot/blob/8a1e61660a0ea2b137b8d750349b7242e5b25ef1/bp_me/src/v/cce/bp_cce_msg.v#L179

Currently, we track only outstanding uncached requests. We should instead increment on mem_cmd_v_o and decrement on mem_resp_yumi_o, using bsg_flow_counter

muwyse, 2020-07-22:
Agreed. Using bsg_flow_counter to track mem_cmd/resp send and receive in bp_cce_msg takes care of most of this issue.

The other small change required is to stall the ucode engine if mem_cmd_ready_i is high but the credit counter has no remaining credits, and force the ucode engine to wait until a memory response returns and is processed.

muwyse, 2020-08-13:
#606 takes care of this for uncached mode. A second PR will fix for normal mode.

muwyse, 2020-08-13:
#607 finishes changes.

PR #606 (2020-08-13): add memory credit counter to ucode CCE
This PR closes #581 by adding a memory credit counter to the ucode CCE. It also fixes the transition from uncached only to normal operation mode.
closed 2020-08-13 by muwyse