black-parrot/black-parrot #604: Remove Zachjs

taylor-bsg, 2020-08-08:


dpetrisko, 2020-08-08:
It's fair since the tool can't currently translate BP. The shame is that the other alternative, bsg_sv2v, requires a DC license whereas zachjs-sv2v does not so it would be nice to someday include it.

What's the general rule here? Only include toolflows which are fully compatible with BP?

taylor-bsg, 2020-08-08:
Yep

zachjs, 2020-08-14:
I'm working through some of the conversion issues. I don't think sv2v is far off from supporting all of the features used by this project. I do have a few questions, though.

Within `bp_be_calculator_top.v`, there is a reference to a member `fmatype` which is not part of the union `rv64_instr_s`. The struct/union conversion raises an error due to the invalid access. While the block of code is disabled with a parameter, sv2v cannot reasonably assume that the module will not be instantiated that flag flipped. I'm not sure what the best behavior here is. Would it be reasonable for the source to be changed in this instance?

In `bp_clint.v`, there appears to be a malformed structure literal.
```SystemVerilog
assign mem_resp_lo =
  '{msg_type       : mem_cmd_li.msg_type
    ,addr          : mem_cmd_li.addr
    ,payload       : mem_cmd_li.payload
    ,size          : mem_cmd_li.size
    ,data          : cce_block_width_p'(rdata_lo)
    };
```
`mem_resp_lo` is of type `bp_cce_mem_msg_s`. The relevant struct definitions are:
```SystemVerilog
typedef struct packed
{
  bp_cce_mem_msg_payload_s                     payload;
  bp_mem_msg_size_e                            size;
  logic [addr_width_mp-1:0]                    addr;
  bp_cce_mem_cmd_type_e                        msg_type;
} bp_cce_mem_msg_header_s;

typedef struct packed
{
  logic [data_width_mp-1:0]                    data;
  bp_cce_mem_msg_header_s                      header;
} bp_cce_mem_msg_s;
```
However, I believe the structure literal in `bp_clint.v` is malformed according to Section 10.9.2 of IEEE 1800-2017 (emphasis mine):
> A `member:value` specifies an explicit value for a named member of the structure. **The named member shall be at the top level of the structure**; a member with the same name in some level of substructure shall not be set. The value shall be castable to the member type and is evaluated in the context of an assignment to the named member; otherwise, an error is generated.

Interestingly, a similar structure literal of the same type in `bp_clint_slice.v` appears to be well-formed.

There are also fields access in `bp_l2e_tile.v` and `bp_clint.v` which appear to be missing an indirection through `.header`, such as: `wire local_cmd_li    = (cce_mem_cmd_lo.addr < dram_base_addr_gp);` `cce_mem_cmd_lo`, too, is of type `bp_cce_mem_msg_s`. Am I missing something here?

dpetrisko, 2020-08-14:
Thanks for the report!

> Within bp_be_calculator_top.v, there is a reference to a member fmatype which is not part of the union rv64_instr_s. The struct/union conversion raises an error due to the invalid access. While the [...]

Good question, I think it's reasonable to say the source should be changed in that instance. This particular issue has already been fixed on dev, by the way. Sorting through issues, you may want to start there: https://github.com/black-parrot/black-parrot/blob/dev/bp_common/src/include/bp_common_rv64_pkg.vh#L90

> However, I believe the structure literal in bp_clint.v is malformed according to Section 10.9.2 of IEEE 1800-2017 (emphasis mine):
> bp_l2e_tile.v

Neither bp_clint nor bp_l2e_tile are currently used in any default configuration.  And as they say, untested code is broken code :) You should be able to safely remove these from the flist for testing.

taylor-bsg, 2020-08-14:
Maybe add a comment for those bugs to bp_clint or file an issue?

dpetrisko, 2020-11-20:
Removed, so closing this and opening #682

closed 2020-11-21 by dpetrisko