black-parrot/black-parrot #608: Error while running Tire Kick test

rashmi2383, 2020-08-16:
When running the command **make -C bp_top/syn tire_kick** following error is received:
%Error: prog.nbf:0: $readmem file not found
Aborting...
Aborted (core dumped)
/home/xxx/black-parrot/bp_top/test/tb/bp_tethered/Makefile.verilator:86: recipe for target '/home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.sim.bp_tests.hello_world/tire_kick' failed
make: *** [/home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.sim.bp_tests.hello_world/tire_kick] Error 134

**Looks like it is because of the following failure:**
make[2]: Leaving directory '/home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.build/obj_dir'
/home/xxx/black-parrot/bp_common/syn/Makefile.verilator:61: recipe for target '/home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.build/obj_dir' failed

dpetrisko, 2020-08-17:
Hi, sorry about that.

The fix (already applied on dev) is here: https://github.com/black-parrot/black-parrot/pull/601

dpetrisko, 2020-08-17:
>parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.build/obj_dir'
/home/xxx/black-parrot/bp_common/syn/Makefile.verilator:61: recipe for target

Although, this looks a little strange, possibly a different error. Makefile.verilator:61 is the actual verilate command, which shouldn't be failing. I just recloned and tested on dev, and it's working on my CentOS machine.

What's your environment? What's the Verilator error that you're encountering on your machine?

rashmi2383, 2020-08-17:
I have ubuntu 18.04. I just cloned the repo and did make prep_lite which ran fine. I don't see any other errors except the one I shared.

dpetrisko, 2020-08-17:
>/home/xxx/black-parrot/bp_common/syn/Makefile.verilator:61: recipe for target '/home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.build/obj_dir' failed

There must be a root failure which is causing this makefile failure.  Could you run ```make -n -C  bp_top/syn tire_kick``` to find the verilator command (should fit this template: https://github.com/black-parrot/black-parrot/blob/dev/bp_common/syn/Makefile.verilator#L61) and try running the verilator command bare?

rashmi2383, 2020-08-17:
Doing that, I see this error message:
./simsc: error while loading shared libraries: libsystemc-2.3.3.so: cannot open shared object file: No such file or directory

dpetrisko, 2020-08-17:
That's the execution command.  The verilate command should start with ```cd x; verilator```, not ./simsc

dpetrisko, 2020-08-17:
Hm, I don't understand how you are able to run the simsc. The step where it's being built is failing according to this error message: ```/home/xxx/black-parrot/bp_common/syn/Makefile.verilator:61: recipe for target '/home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.build/obj_dir' failed```

rashmi2383, 2020-08-17:
Here is the output of running **make -n -C bp_top/syn tire_kick**:

make: Entering directory '/home/xxx/black-parrot/bp_top/syn'
make build.sc
make[1]: Entering directory '/home/xxx/black-parrot/bp_top/syn'
make[1]: Nothing to be done for 'build.sc'.
make[1]: Leaving directory '/home/xxx/black-parrot/bp_top/syn'
make /home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.sim.bp_tests.hello_world/simsc
make[1]: Entering directory '/home/xxx/black-parrot/bp_top/syn'
make[1]: '/home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.sim.bp_tests.hello_world/simsc' is up to date.
make[1]: Leaving directory '/home/xxx/black-parrot/bp_top/syn'
cp /home/xxx/black-parrot/bp_common/test/mem/hello_world.mem  /home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.sim.bp_tests.hello_world/prog.mem
cp /home/rashmi/black-parrot/bp_common/test/mem/hello_world.dump /home/rashmi/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.sim.bp_tests.hello_world/prog.dump
cp /home/xxx/black-parrot/bp_me/src/asm/roms/mesi.mem /home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.sim.bp_tests.hello_world/cce_ucode.mem
cd /home/xxx/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.sim.bp_tests.hello_world; ./simsc
make: Leaving directory '/home/xxx/black-parrot/bp_top/syn'

dpetrisko, 2020-08-17:
> make[1]: Nothing to be done for 'build.sc'.

Try ```make -C bp_top/syn clean``` first.  It looks like it thinks the verilator build was successful, so it's not trying it again

dpetrisko, 2020-08-17:
Based on this log, it seems that you're running on master? You'll either need to switch to dev or cherry-pick that fix I posted earlier.  It's missing this line: https://github.com/black-parrot/black-parrot/blob/dev/bp_top/test/tb/bp_tethered/Makefile.verilator#L104

rashmi2383, 2020-08-17:
Ok, I will give it a try.

rashmi2383, 2020-08-17:
That worked.

closed 2020-08-17 by rashmi2383

dpetrisko, 2020-08-17:
Cool!