black-parrot/black-parrot #609: Issue trying to use docker to set up Black Parrot on windows 10

MargotBauman, 2020-08-17:
Following the command **docker-compose up -d**, the following error message comes up:

Removing intermediate container xxx
 ---> xxx
Step 9/12 : RUN useradd --no-log-init --uid $USER_ID --gid $GROUP_ID --shell /bin/bash --create-home build 
 ---> Running in xxx
useradd: invalid user ID '--gid'
ERROR: Service 'bp' failed to build: The command '/bin/sh -c useradd --no-log-init GROUP_ID --shell /bin/bash --create-home build' returned a non-zero code: 3

This looks like it is related to command syntax in the **dockerfile**, which uses $(id -u) and $(id -g) syntax. "id" is not a valid command in windows (equivalent of id -u would be whoami /user & id -g would be whoami /groups).

dpetrisko, 2020-08-17:
The docker instructions have only been tested on MacOS.  If you get a windows variant working, we'd appreciate a PR to the README!

dpetrisko, 2022-07-30:
Resolving for lack of action. Would appreciate anyone on windows trying out the docker flow in black-parrot-sim

closed 2022-07-30 by dpetrisko