black-parrot/black-parrot #647: No results displayed while running tire_kick

benleone90, 2020-10-23:
After reviewing the issue mentioned in #608 I cloned the dev branch to my local machine (Ubuntu 20.04) and ran `make prep_lite` which seemed to run fine. When running `make -C bp_top/syn tire_kick` I received this output:

Hello world!
[CORE0 FSH] PASS
All cores finished! Terminating...
\- /home/xxx/Desktop/black-parrot/bp_top/test/common/bp_nonsynth_host.v:202: Verilog $finish
Finishing test
Exiting
[CORE0 FSH] PASS
make: [/home/xxx/Desktop/black-parrot/bp_top/test/tb/bp_tethered/Makefile.verilator:112: /home/xxx/Desktop/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_unicore_cfg.none.sim.bp_tests.hello_world/tire_kick] Error 1 (ignored)
make: Leaving directory '/home/benleone/Desktop/black-parrot/bp_top/syn'

FWIW, I ran this earlier this week and received the correct output listed in the documentation.

dpetrisko, 2020-10-23:
Hi, thanks for the report! There was a change #587 which made reports print on the “final” event, rather than manually. Adding this line should fix it:

https://github.com/black-parrot/black-parrot/blob/dev_fix_nbfconfigload/bp_top/test/tb/bp_tethered/test_bp.cpp#L63

benleone90, 2020-10-23:
Thanks! This fix worked after adding the line mentioned, running `make -C bp_top/syn clean` and then running `make -C bp_top/syn tire_kick` again, I got the output.

dpetrisko, 2020-10-23:
Great! I’m going to keep this issue open for a bit as a reminder to merge in the fix

dpetrisko, 2020-10-29:
Fix merged into dev

closed 2020-10-29 by dpetrisko