black-parrot/black-parrot #653: some of the beebs programs don't compile, with a "no rule to make target -lm"

hchenji, 2020-10-29:
This is on Centos 7 with devtoolset-8 when I run make progs in black-parrot. In bp_common/test, the beebs repo and included Makefiles seem to have issues when compiling the cubic, nbody, st, stb_perlin, whetstone, wikisort programs.

I've traced it down to Makefile.in in the individual programs' folders in the beebs repo. the cubic_DEPENDENCIES variable for e.g. includes $(LIBM), which is set to -lm since the riscv gcc toolchain supplies libmath. However this is not a dependency as there is no rule to create a -lm file. Removing LIBM from the variable in the relevant Makefile.in fixes it. 

Or...am I missing something? is there any other way to fix this? Maybe a different devtoolset?

dpetrisko, 2020-10-29:
Hello, are there any errors when compiling the riscv-gnu-toolchain through make prep? I believe the math libraries should be compiled as part of riscv_newlib.

@farzamgl any ideas?

hchenji, 2020-10-29:
No problems with the toolchain. Even the other test suites like riscv_tests compile and work. It's just those particular programs in beebs - the other tests like ctl_vector work fine.

dpetrisko, 2020-10-29:
Yeah, looks like it's written that way to support a (stubbed?) version of libm, which may be a true dependency.  Curious that it doesn't show up on my system, must be some subtly different tool version.  I think the fix is remove LIBM as you say

farzamgl, 2020-10-29:
> @farzamgl any ideas?
Not really. I've never seen this issue with beebs. It's probably because of different tool versions.

dpetrisko, 2020-10-29:
@hchenji can you try #654 and let me know if it works for you?

hchenji, 2020-10-30:
#654 works.

here's a working example to reproduce

[docker.zip](https://github.com/black-parrot/black-parrot/files/5462610/docker.zip)

mkdir bp
//cd and put Dockerfile and compose here, checkout black-parrot
docker-compose run bp
scl enable devtoolset-7 bash // happens with devtoolset-8 too
make prep_lite -j
make prep -j

`make[6]: Entering directory '/root/black-parrot/bp_common/test/src/beebs/src/cubic'
  CC       libcubic.lo
  CC       basicmath_small.lo
  CCLD     libcubic.la
make[6]: *** No rule to make target '-lm', needed by 'cubic'.  Stop.
`

dpetrisko, 2020-10-30:
Thanks for the report! Fix merged in dev

closed 2020-10-30 by dpetrisko