black-parrot/black-parrot #675: Few ISA tests did not exit simulation with reduced L1 dcache sets

sammy17, 2020-11-12:
rv64ua-p-amo*, rv64ua-p-lrsc and rv64ui-v-* ISA tests never exited the simulation with the reduced cache size (**dcache_sets=32 and dcache_sets=16**). 

These tests ran without any issue with the default cache size.

With reduced cache size, tests seem to be in a loop (processor was running).

dpetrisko, 2020-11-20:
@Shashank-Vijay can you take a look at this?

dpetrisko, 2020-11-20:
Hello, thanks for the issue! Because these are a VIPT cache, we've made an assumption that configurations will have index+offset bits == page_offset_bits. In order to reduce size, you must instead reduce associativity.  The configurations in aviary for varying L1 cache size obey this constraint.  Note there's no technical reason that the cache could have index+offset_bits < page_offset_bits, but that would require one of the page_offset bits bleeding into the tag check, which will complicate the code. 

So for instance, 
```
512 block_width, 64 sets, 8-way == 32kB
512 block_width, 64 sets, 4-way == 16kB
256 block_width, 128 sets, 4-way == 16kB
128 block_width, 256 sets, 2-way == 8kB
```
are all legal configurations

I've added this assumption explicitly in #683

closed 2020-12-22 by dpetrisko