black-parrot/black-parrot #682: Support zachjs-sv2v

dpetrisko, 2020-11-20:
https://github.com/zachjs/sv2v

Fully open-source systemverilog to verilog converter.  There are currently some bugs in the tools and bugs in BP preventing this integration.  Mostly we just need someone to put in the time to sorting out the rough edges of the spec.

developandplay, 2021-04-03:
I'll look into this a bit on the side.
Afterall it would be really nice to have yosys support for blackparrot and run it on my lattice board.

dpetrisko, 2021-04-04:
Generally we use https://github.com/bespoke-silicon-group/bsg_sv2v to convert to a single 'pickled' .v file.  We can send you a file to work with if you'd like, the long-term plan is to publish them along with major releases of BP.

It'd be great to have this as well, I'll just warn you it's non-trivial

developandplay, 2021-04-04:
Oh yes I'd be happy to look into one of the pickled files until I figure out the licenses...
Releasing the verilog files sounds like a great idea!

Yea I looked into Zach's project a bit and exotic SystemVerilog features or BP bugs are probably not my strength (yet)...
That being said I did work on some programming language translation project in Haskell so the sv2v project structure seems somewhat familiar.
Since Zach seems to be very active regarding his project...maybe we can make some progress.

closed 2021-12-16 by taylor-bsg