black-parrot/black-parrot #694: Verilator checkpoint flow internally segfaults

dpetrisko, 2020-11-29:
```	
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed PATH/external/bin/verilator_bin --sc -O1 --x-assign fast --x-initial fast --threads 1 --top-module testbench -f flist.vcs config.vlt --build --exe -o simsc -Wno-timescalemod --trace --trace-structs --trace-depth 15 --Wno-fatal --Wno-lint --Wno-style --Wno-widthconcat --Wno-unoptflat -CFLAGS -std\=c\+\+14 -IPATH/external/share/verilator/include/vltstd/ -LDFLAGS -LPATH/external/lib\ -lpthread\ -ldramsim3\ -ldromajo_cosim\ -Wl\,-rpath\=PATH/external/lib -pvalue\+cce_trace_p\=0 -pvalue\+lce_trace_p\=0 -pvalue\+cmt_trace_p\=0 -pvalue\+dram_trace_p\=0 -pvalue\+icache_trace_p\=0 -pvalue\+dcache_trace_p\=0 -pvalue\+vm_trace_p\=0 -pvalue\+core_profile_p\=0 -pvalue\+pc_profile_p\=0 -pvalue\+br_profile_p\=0 -pvalue\+checkpoint_p\=1 -pvalue\+cosim_p\=1 -pvalue\+cosim_memsize_p\=128 -pvalue\+cosim_instr_p\=0 -pvalue\+warmup_instr_p\=0 -pvalue\+preload_mem_p\=0 -pvalue\+use_ddr_p\=0 -pvalue\+use_dramsim3_p\=0 -pvalue\+dram_fixed_latency_p\=10 \+define\+BP_SIM_CLK_PERIOD\=5000 \+define\+dram_pkg\=bp_dramsim3_lpddr_2Gb_x16_pkg \+define\+den2048Mb\+sg5\+x16\+FULL_MEM
```

Works on VCS, so seems to be a verilator bug.  Should try out different compilation flags

farzamgl, 2021-01-03:
I think the checkpoint flow currently works on sw_dev. I haven't seen the segfault issue before though, can one of you verify if the sw_dev changes fix the issue?

dpetrisko, 2021-01-04:
Try re-enabling in CI, if it works there we’re good

closed 2021-01-05 by farzamgl