black-parrot/black-parrot #701: xxd dependency

getziadz, 2020-12-08:
I encountered a dependency on `xxd` (vim-common package) which was not specified anywhere.
Though trivial to resolve, it may be beneficial to create a section where you list dependecies for getting started with BlackParrot.
The section could include a description of systems that the BlackParrot infrastructure has been tested.

dpetrisko, 2020-12-08:
You can find this info here in the getting started guide:

https://github.com/black-parrot/black-parrot/blob/master/docs/getting_started.md#prerequisites

Would happily accept a PR for additional Fedora requirements!

getziadz, 2020-12-08:
The CentOS entry should cover Fedora and RHEL distributions. I don't see vim-common on the prerequisites list but it may be pulled from one of the other packages or it could be that the standard distribution comes with it preinstalled.

dpetrisko, 2020-12-08:
Yeah, I'm not sure.  The CentOS entry is the same as the Dockerfile I use so it should be a clean install, but I always include vim before adding the dependencies. Adding vim-common to the list is reasonable a safety net

PR #731 (2021-01-12): Updating dependency list
fixes #701
closed 2021-01-12 by dpetrisko