black-parrot/black-parrot #714: package should actually be include file

taylor-bsg, 2020-12-22:
https://github.com/black-parrot/black-parrot/blob/4e829d9a91786bbf99af03fc9d1837540189d4f7/bp_me/src/include/v/bp_cce_pkg.v#L10

packages do not contain macros, so this works just because the order of systemverilog files is being specified in a way that this should be read in. should be a .vh file?

dpetrisko, 2020-12-22:
Yeah, it’s a hack that seems to work decently well across tools. Packages are always first in our flists, so they are elaborated first. When they’re elaborated first, the macros are globally defined before any .sv files are compiled. 

It may complicated things to mix between packages and includes. Is there a tool where this approach is causing problems?

taylor-bsg, 2020-12-23:
See slack message I sent you where people were having problems. In general
requiring people to order flists inhibits adoption and increases time to
get things setup. The fix (include files) is relatively simple and mirrors
our bsg_defines change.

taylor-bsg, 2020-12-23:
It has to do with differences between what tools include in a compilation
unit. Synopsys puts everything in one unit by default which is not the
default in other tools.

dpetrisko, 2020-12-23:
Are there other consequences to adopting a compilation-unit-per-file rather than single-compilation-unit methodology?

taylor-bsg, 2020-12-23:
Mostly in testbenches; IIRC hierarchical naming works differently.

PR #737 (2021-01-26): Breaking reliance on `define order
Fixes #714 #715

Should improve tool compatibility, no longer relies on compile order.

- All defines go into top level defines: common_defines, fe_defines, be_defines, me_defines, subdefines are `included there
- Only bp_common_pkg, bp_fe_pkg, bp_be_pkg, bp_me_pkg top level packages remain
- What was before a subpackage is now a "_pkgdef" or a set of parameters and typedefs which need to be `included into a "_pkg"
- ~It's possible we'll also need to at least `include bp_common_defines on every file~ 
- At least verilator and vcs are fine with this right now. [... cut]
closed 2021-01-28 by dpetrisko