black-parrot/black-parrot #718: Support non-blocking loads

dpetrisko, 2020-12-23:
We already support separate commit / writeback. Adding a simple in-order scoreboard should allow us to track dependencies for these long-latency instructions.

Open Questions:
- Do we support non-point-to-point loads?
- If so, how do we track the requests? Adding reg_id payload?
- Does cache engine or cache track the outstanding requests?
- Does this follow through the network? Or do we track the return address?

taylor-bsg, 2020-12-23:
The manycore is a good guide for how this should be done..

dpetrisko, 2020-12-23:
We'll definitely look at the manycore. The scoreboarding approach will most likely be identical.  However, there are distinct differences in the system e.g. cache coherence, caches at all ..., that complicate the implementation

dpetrisko, 2023-07-18:
Closing; supported

closed 2023-07-18 by dpetrisko