black-parrot/black-parrot #723: BHT and BTB should have init_done signals

sripathi-muralitharan, 2020-12-26:
The BHT and BTB are memories that have their own clear sequence at the initialization. However, it is possible that the core starts executing before this sequence is completed since it only depends on the freeze signal from the config bus. 

The core should start after all initialization sequences are complete. Having init_done signals on the BHT and BTB should solve the problem. Maybe accept the initial FE cmd from the director only if this initialization is complete? 

This was observed when debugging issue #700  (i.e simulating the core with a no L2 configuration in which case the NBF boot sequence is completed faster). 

To reproduce this problem - 
1. On the me_dev branch switch to commit no - dd4ba8c868c4625d5e9e8053e6f4d7e2b03a5e80
2. Run the rv64ui-v-lui program with e_bp_unicore_no_l2_cfg (Actually all programs (observed for a v test that works and one that doesn't work) with the no L2 cfg have this problem but it gets masked somehow)

Note: The current version of the me_dev branch seems to mask this problem.

dpetrisko, 2021-01-12:
I'll look into this while I investigate the stalling front-end changes

dpetrisko, 2021-02-25:
#773

dpetrisko, 2021-03-16:
Fixed in dev

closed 2021-03-16 by dpetrisko