black-parrot/black-parrot #725: BedRock macros are strongly coupled

dpetrisko, 2020-12-28:
If you try to use  

```
`declare_bp_bedrock_mem_if(paddr_width_p, uce_mem_data_width_lp, lce_id_width_p, lce_assoc_p, uce);
```
without 
```
`declare_bp_bedrock_mem_if_widths(paddr_width_p, uce_mem_data_width_lp, lce_id_width_p, lce_assoc_p, uce)
```
it will complain of a missing payload width declaration.  To fix, can either:

1) pass in the payload_width_lp as a parameter
2) recalculate payload width internally in declare_bp_bedrock_mem_if, using ____payload_width_lp = to obfuscate the name.

muwyse, 2021-02-05:
@dpetrisko can you take a look at this commit: https://github.com/black-parrot/black-parrot/commit/c48fe7dfeac7647c3ac217498f5ef2c6e2364c2a

I think it fixes the problem (the ME testbench compiled and ran without issue using vcs)

dpetrisko, 2021-02-05:
Perfect!

commit c48fe7df (2021-02-05): decouple bedrock width and declaration macros; fixes #725
closed 2021-02-25 by dpetrisko