black-parrot/black-parrot #728: Maintain FPGA Compatibility

farzamgl, 2021-01-08:
Currently, some bare-metal tests have issues running on the Xilinx Virtex Utralscale+ board. We have to debug these issues using Vivado's post-synth simulation and add compatibility tests to our regression suite. (example: https://github.com/black-parrot/black-parrot/pull/706)

dpetrisko, 2021-01-08:
@Shashank-Vijay is working on post-synth vivado sims, not with the Xilinx IPs though. The bug in #706 would have been caught without the IPs, but I can imagine bugs which would not

taylor-bsg, 2021-01-08:
And to be clear, we are okay with Vivado synthesis and VCS simulation.
Vivado is 10X slower than VCS.

dpetrisko, 2021-01-08:
>And to be clear, we are okay with Vivado synthesis and VCS simulation.
>Vivado is 10X slower than VCS.

That's interesting! Probably want to support both, but run most regressions with VCS for speed

taylor-bsg, 2021-01-08:
Just in terms of spending time on things, bigger fish to fry if it takes a
lot of work to get Vivado sim going -- people can use Verilator

dpetrisko, 2021-01-08:
>Just in terms of spending time on things, bigger fish to fry if it takes a
lot of work to get Vivado sim going -- people can use Verilator

True, although one major use case is students working with just Vivado for FPGA-based classes. In any case, the idea is for the makefile flow to support orthogonal synth and sim by switching out simulation flists, so it shouldn't be too bad.  But point taken, won't spend an inordinate amount of time there

dpetrisko, 2021-01-09:
>Just in terms of spending time on things, bigger fish to fry if it takes a
lot of work to get Vivado sim going -- people can use Verilator

From talking to Dustin and Jon, not worth supporting vivado sim.  Too many bugs!

taylor-bsg, 2021-01-10:
Yep. There is one use case though that is not covered by Verilator, which
is if the Xilinx IP blocks are encrypted. But cross that bridge later.

M