black-parrot/black-parrot #736: Support Iterative Filling in Direct Mapped Cache Configurations

sripathi-muralitharan, 2021-01-24:
Currently, the direct-mapped caches are capable of supporting fills of size 512. This means that the downstream serialization/deserialization/buffering logic would have to be 512-bit wide. 

We should support iterative filling in direct-mapped caches because - 
1. We are not fully parameterizable until we do.
2. It has a large area overhead which is not conducive to creating a Tiny BlackParrot.

One possible fix could be to bank the data memories to have 8 banks of 64-bit widths instead of 1 bank with 512-bit width. Modifying the banking scheme this way also has the benefit of creating square memories which have good area and power properties.

dpetrisko, 2021-01-24:
Disagree on the shape of the rams being better for 64x64, they’re pretty well optimized at 512x64. If we really wanted to bank them, we could physically bank the hardened swap rams. 

I agree that it doesn’t make much sense to have a 512-bit cache line with a direct-mapped cache. Parameterization is good for supporting different use-cases, but parameterization for parameterization’s sake just makes it harder to maintain. Shrinking the line size and matching the fill size is better for TinyParrot, no? That would minimize the number of data registers in the cache. 

There’s a prefetching argument for keeping to say, 128-bit lines rather than 64. But that’s obviously workload dependent and you could effectively do the same thing with a simple next-line prefetcher. 

In general, I don’t hate the idea. It would require modifying the Cache Engine Interface. But the hardware cost of changing the fill mask to a sub-bank mask is not terrible, assuming we enforce alignment of sub-banks.

sripathi-muralitharan, 2021-01-24:
Shrinking the line size and matching to fill size would be fine from an area perspective (provided we physically depth-bank the tall memory if we still want a 4KB cache) but will have a large performance hit based on workload. Given that the tiny core is expected to be Linux capable, can't we assume the workloads might be such that a fill width = line width cache makes the core terrible? (Not sure what other tiny linux capable cores have...)

As I see it now, we are restricted to have bank widths to be atleast fill width size and having sub-block filling would be pretty useful if we want to take full advantage of the stream and burst protocols right?

dpetrisko, 2021-01-24:
> Shrinking the line size and matching to fill size would be fine from an area perspective

Yeah, the exercise is to make the tiniest BlackParrot possible. Performance tuning comes after we profile the workload.
Analysis of VM-heavy workloads is tricky (and we don't have any set up, other than boot which is not representative). In many workloads, most of the time is spent in PTW, which is fully determined by how much of the PT is resident in the cache (since it's traversing a linked list and has no locality). Having twice of big of a line size at absolute most give you 2x fewer cache misses (in reality it's much smaller, especially for data cache).

Note that doing partial fill also slows down the fill, writeback and notably coherence operations by a several cycles apiece. This is most likely not noticeable in when attached to a slow DRAM device, but with an L2 can be a significant hit to latency-sensitive workloads.

> As I see it now, we are restricted to have bank widths to be atleast fill width size and having sub-block filling would be pretty useful if we want to take full advantage of the stream and burst protocols right?

There's not really much point to having a width smaller than your L2 cache word size, or absent L2, your data bus size. For our chips, we usually serialize to off-chip so the size is mostly irrelevant. The L2 is comfortable at 64-bits, but is still pretty efficient at 32-bits.  For attaching to preexisting IP, AXI-buses typically support 64 bits, although if only AXI-lite IP is available, it may be necessary to downshift to 32 bits. I'm not sure what Wishbone buses support.

So, I think it's a fine change in the context of supporting additional off-chip buses. I don't think that it's necessary for analysis of TinyParrot.

dpetrisko, 2026-09-08:
Closing for unclear benefits

closed 2026-09-08 by dpetrisko