black-parrot/black-parrot #757: Gettin Started Full fails on Ubuntu

gabaxter, 2021-02-11:
Hi, 

Following the steps of [https://github.com/black-parrot/black-parrot/blob/master/docs/getting_started.md](https://github.com/black-parrot/black-parrot/blob/master/docs/getting_started.md),  namely:

**Prereq:**

`sudo apt-get install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev wget byacc device-tree-compiler python gtkwave`

**Download/build:**
`# Clone the latest repo`
`git clone https://github.com/black-parrot/black-parrot.git`
`cd black-parrot`
`# make prep is a meta-target which will build the RISC-V toolchains, programs and microcode`
`#   needed for a full BlackParrot evaluation setup.`
`# Users who are changing code can use the 'libs' 'progs' or 'ucode' targets as appropriate`
`# For faster builds, make prep -j is parallelizable!`
`make prep`

**On Ubuntu 18.04.5v LTS fails after ~1H of build time fails with these messages:**
`riscv64-unknown-linux-gnu-g++  links-dso-program.cc -c -I/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/ -isystem /home/glenn/work/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/include/c++/9.2.0 -isystem /home/glenn/work/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/include/c++/9.2.0/riscv64-unknown-linux-gnu -isystem /home/glenn/work/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/include/c++/9.2.0/backward -mcmodel=medany -g -O2 -Wall -Wwrite-strings -Wundef -fmerge-all-constants -frounding-math -fno-stack-protector          -I../include -I/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/support  -I/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64  -I../sysdeps/unix/sysv/linux/riscv/rv64  -I../sysdeps/unix/sysv/linux/riscv  -I../sysdeps/riscv/nptl  -I../sysdeps/unix/sysv/linux/generic  -I../sysdeps/unix/sysv/linux/wordsize-64  -I../sysdeps/unix/sysv/linux/include -I../sysdeps/unix/sysv/linux  -I../sysdeps/nptl  -I../sysdeps/pthread  -I../sysdeps/gnu  -I../sysdeps/unix/inet  -I../sysdeps/unix/sysv  -I../sysdeps/unix  -I../sysdeps/posix  -I../sysdeps/riscv/rv64/rvd  -I../sysdeps/riscv/rv64/rvf  -I../sysdeps/riscv/rvf  -I../sysdeps/riscv/rvd  -I../sysdeps/riscv/rv64  -I../sysdeps/riscv  -I../sysdeps/init_array  -I../sysdeps/ieee754/ldbl-128  -I../sysdeps/ieee754/dbl-64  -I../sysdeps/ieee754/flt-32  -I../sysdeps/wordsize-64  -I../sysdeps/ieee754  -I../sysdeps/generic  -I.. -I../libio -I. -nostdinc -isystem /home/glenn/work/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/include -isystem /home/glenn/work/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/include-fixed -isystem /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/linux-headers/include  -D_LIBC_REENTRANT -include /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/libc-modules.h -DMODULE_NAME=nonlib -include ../include/libc-symbols.h       -DTOP_NAMESPACE=glibc -o /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/support/links-dso-program.o -MD -MP -MF /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/support/links-dso-program.o.dt -MT /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/support/links-dso-program.o
riscv64-unknown-linux-gnu-gcc  -nostdlib -nostartfiles -o /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/support/links-dso-program    -Wl,-z,combreloc -Wl,-z,relro -Wl,--hash-style=both /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/csu/crt1.o /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/csu/crti.o `riscv64-unknown-linux-gnu-gcc   --print-file-name=crtbegin.o` /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/support/links-dso-program.o -lstdc++ -lgcc -lgcc_s  -Wl,-dynamic-linker=/lib/ld-linux-riscv64-lp64.so.1 -Wl,-rpath-link=/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/math:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/dlfcn:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/nss:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/nis:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/rt:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/resolv:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/mathvec:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/support:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/crypt:/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/nptl /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/libc.so.6 /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/libc_nonshared.a -Wl,--as-needed /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/ld.so -Wl,--no-as-needed -lgcc  `riscv64-unknown-linux-gnu-gcc   --print-file-name=crtend.o` /home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/csu/crtn.o
/home/glenn/work/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lstdc++
/home/glenn/work/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lgcc_s
collect2: error: ld returned 1 exit status
../Rules:193: recipe for target '/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/support/links-dso-program' failed
make[7]: *** [/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/support/links-dso-program] Error 1
make[7]: Leaving directory '/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/support'
Makefile:258: recipe for target 'support/others' failed
make[6]: *** [support/others] Error 2
make[6]: Leaving directory '/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/riscv-glibc'
Makefile:9: recipe for target 'all' failed
make[5]: *** [all] Error 2
make[5]: Leaving directory '/home/glenn/work/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64'
Makefile:234: recipe for target 'stamps/build-glibc-linux-rv64imafd-lp64' failed
make[4]: *** [stamps/build-glibc-linux-rv64imafd-lp64] Error 2
make[4]: Leaving directory '/home/glenn/work/black-parrot/external/riscv-gnu-toolchain'
/home/glenn/work/black-parrot/external/Makefile.tools:46: recipe for target 'gnu_build' failed
make[3]: *** [gnu_build] Error 2
make[3]: Leaving directory '/home/glenn/work/black-parrot'
/home/glenn/work/black-parrot/external/Makefile.tools:82: recipe for target '/home/glenn/work/black-parrot/external/touchfiles/gnu.70063d751135fb22e754ff14bc969ffe9e905fbd' failed
make[2]: *** [/home/glenn/work/black-parrot/external/touchfiles/gnu.70063d751135fb22e754ff14bc969ffe9e905fbd] Error 2
make[2]: Leaving directory '/home/glenn/work/black-parrot'
Makefile:62: recipe for target 'tools' failed
make[1]: *** [tools] Error 2
make[1]: Leaving directory '/home/glenn/work/black-parrot'
Makefile:24: recipe for target 'prep' failed
make: *** [prep] Error 2
`

If someone can help figure out this issue, I will be happy to test and verify.  I am totally new too BP so I do not even know were to start looking.   It looks like an error in a ENV Var.

Thank you!

taylor-bsg, 2021-02-11:
Maybe this will help:

https://askubuntu.com/questions/346377/cannot-find-lgcc-s

muwyse, 2021-02-11:
It seems odd that running the prereq step would leave the system in a state that it doesn't know about libgcc. That post is helpful for determining if ldconfig knows about the path, but I'm not sure it gives a user-friendly solution.

Perhaps explicitly installing a relatively recent version of gcc and g++ and making those the default versions would resolve?

gabaxter, 2021-02-11:
Thank you for the quick response!

`glenn@xxxx:~/work/black-parrot/external(master)$ cd /home/glenn/work/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/`
`glenn@xxxx:~/work/black-parrot/external/riscv64-unknown-linux-gnu/bin(master)$ ls`
`ar  as  ld  ld.bfd  nm  objcopy  objdump  ranlib  readelf  strip`

Those are the tools installed at the location.  The repo is designed to self get everything, so it is not referring to prerequisite tools since those will be in `/usr/bin` or `/usr/local/bin`.  

I was concerned it generated correct paths, but in looking at it, it seems to have done so.   Not sure what else I can offer.  Let me know if you want to see any particular system path.

muwyse, 2021-02-12:
Check out the apt-get install command in the Dockerfile we provide (which launches an Ubuntu 18.04 image). https://github.com/black-parrot/black-parrot/blob/master/Dockerfile

From a brief scan, it looks like it installs a couple extra things than are listed in the getting started guide. To me it looks like the error is occurring while building the RISC-V gcc toolchain submodule, so my best guess is a missing dependency for the gcc build. You could also try running the prereq installs listed on the toolchain repo (https://github.com/riscv/riscv-gnu-toolchain) then rebuilding.

If you have multiple cores to spare you can use `make prep -jN` where N is some number of cores to do the BP make in parallel (this can significantly speed up build time).

gabaxter, 2021-02-12:
I took the docker apt, and it **is** different from what is in readme.  It looks like `python-yaml` is missing.  Here is the complete output of running the docker apt:

`Reading package lists... Done`
`Building dependency tree       `
`Reading state information... Done`
`Note, selecting 'libexpat1-dev' instead of 'libexpat-dev'`
`autoconf is already the newest version (2.69-11).`
`automake is already the newest version (1:1.15.1-3ubuntu2).`
`autotools-dev is already the newest version (20180224.1).`
`bc is already the newest version (1.07.1-2).`
`bison is already the newest version (2:3.0.4.dfsg-1build1).`
`build-essential is already the newest version (12.4ubuntu1).`
`device-tree-compiler is already the newest version (1.4.5-3).`
`flex is already the newest version (2.6.4-6).`
`gawk is already the newest version (1:4.1.4+dfsg-1build1).`
`libgmp-dev is already the newest version (2:6.1.2+dfsg-2).`
`libmpc-dev is already the newest version (1.1.0-1).`
`libmpfr-dev is already the newest version (4.0.1-1).`
`libtool is already the newest version (2.4.6-2).`
`patchutils is already the newest version (0.3.4-2).`
`pkg-config is already the newest version (0.29.1-0ubuntu2).`
`python is already the newest version (2.7.15~rc1-1).`
`zlib1g-dev is already the newest version (1:1.2.11.dfsg-0ubuntu2).`
`byacc is already the newest version (20140715-1build1).`
`gperf is already the newest version (3.1-1).`
`gtkwave is already the newest version (3.3.86-1).`
`swig is already the newest version (3.0.12-1).`
`texinfo is already the newest version (6.5.0.dfsg.1-2).`
`apt-utils is already the newest version (1.6.12ubuntu0.2).`
`bash is already the newest version (4.4.18-2ubuntu1.2).`
`curl is already the newest version (7.58.0-2ubuntu3.12).`
`gettext-base is already the newest version (0.19.8.1-6ubuntu0.3).`
`git is already the newest version (1:2.17.1-1ubuntu0.7).`
`libexpat1-dev is already the newest version (2.2.5-3ubuntu0.2).`
`python3-dev is already the newest version (3.6.7-1~18.04).`
`python3-dev set to manually installed.`
`tzdata is already the newest version (2021a-0ubuntu0.18.04).`
`vim is already the newest version (2:8.0.1453-1ubuntu1.4).`
`wget is already the newest version (1.19.4-1ubuntu2.2).`
`The following package was automatically installed and is no longer required:`
`  libllvm9`
`Use 'sudo apt autoremove' to remove it.`
`The following NEW packages will be installed:`
`  python-yaml`
`0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.`
`Need to get 115 kB of archives.`
`After this operation, 479 kB of additional disk space will be used.`
`Get:1 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 python-yaml amd64 3.12-1build2 [115 kB]`
`Fetched 115 kB in 0s (520 kB/s)     `
`Selecting previously unselected package python-yaml.`
`(Reading database ... 246021 files and directories currently installed.)`
`Preparing to unpack .../python-yaml_3.12-1build2_amd64.deb ...`
`Unpacking python-yaml (3.12-1build2) ...`
`Setting up python-yaml (3.12-1build2) ...`

**I then ran**
`make prep -j 8` 

**After about 20 minutes it made is much farther, but still failed:**

`...`
`pypy3.6-v7.3.2-linux64/lib-python/3/antigravity.py`
`cd /home/glenn/work/black-parrot/external/bsg_sv2v/tools/pypy3; mv pypy3.6-v7.3.2-linux64/* .`
`cd /home/glenn/work/black-parrot/external/bsg_sv2v/tools/pypy3; rmdir pypy3.6-v7.3.2-linux64/`
`mkdir -p /home/glenn/work/black-parrot/external/bsg_sv2v/tools`
`virtualenv -p /home/glenn/work/black-parrot/external/bsg_sv2v/tools/pypy3/bin/pypy3 /home/glenn/work/black-parrot/external/bsg_sv2v/tools/virtual_env`
`/bin/bash: virtualenv: command not found`
`Makefile:144: recipe for target '/home/glenn/work/black-parrot/external/bsg_sv2v/tools/virtual_env' failed`
`make[4]: *** [/home/glenn/work/black-parrot/external/bsg_sv2v/tools/virtual_env] Error 127`
`make[4]: Leaving directory '/home/glenn/work/black-parrot/external/bsg_sv2v'`
`/home/glenn/work/black-parrot/external/Makefile.tools:77: recipe for target 'bsg_sv2v_build' failed`
`make[3]: *** [bsg_sv2v_build] Error 2`
`make[3]: Leaving directory '/home/glenn/work/black-parrot'`
`/home/glenn/work/black-parrot/external/Makefile.tools:86: recipe for target '/home/glenn/work/black-parrot/external/touchfiles/bsg_sv2v.03f1e921ef9319218798a5a8826b79dcc8b19056' failed`
`make[2]: *** [/home/glenn/work/black-parrot/external/touchfiles/bsg_sv2v.03f1e921ef9319218798a5a8826b79dcc8b19056] Error 2`
`make[2]: Leaving directory '/home/glenn/work/black-parrot'`
`Makefile:62: recipe for target 'tools' failed`
`make[1]: *** [tools] Error 2`
`make[1]: Leaving directory '/home/glenn/work/black-parrot'`
`Makefile:24: recipe for target 'prep' failed`
`make: *** [prep] Error 2`

**This in turn looks like another dependency?**
`/bin/bash: virtualenv: command not found`

Thoughts?

gabaxter, 2021-02-12:
On a different but related topic, the docker flow just finally failed as well (after ~8H of run on high end MacBook Pro).  Here is what I believe are the initial location of the error. This is using Docker Desktop on Mac, starting Docker Desktop and then running the docker install commands from the local MacOS shell.  AFAIK, this should work.  If not, please let me know how the docker portion is to run.  

riscv64-unknown-linux-gnu-gcc    -nostdlib -nostartfiles -r -o /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/librtld.os '-Wl,-(' /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/dl-allobjs.os /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/rtld-libc.a -lgcc '-Wl,-)' \
	  -Wl,-Map,/home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/librtld.os.map
riscv64-unknown-linux-gnu-gcc    -nostdlib -nostartfiles -shared -o /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/ld.so.new		\
	  -Wl,-z,combreloc -Wl,-z,relro -Wl,--hash-style=both -Wl,-z,defs 	\
	  /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/librtld.os -Wl,--version-script=/home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/ld.map		\
	  -Wl,-soname=ld-linux-riscv64-lp64.so.1			\
	  -Wl,-defsym=_begin=0
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/librtld.os: in function `process_envvars':
/home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/rtld.c:2493: undefined reference to `__GI___open64_nocancel'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/librtld.os: in function `dl_main':
/home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/rtld.c:1333: undefined reference to `__access'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/rtld.c:2194: undefined reference to `__access'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/librtld.os: in function `is_dst':
/home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/dl-load.c:211: undefined reference to `__GI___close_nocancel'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/librtld.os: in function `is_trusted_path_normalize':
/home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/dl-load.c:128: undefined reference to `__GI___fxstat64'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/dl-load.c:166: undefined reference to `__GI___close_nocancel'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/librtld.os: in function `_dl_map_segments':
/home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/./dl-map-segments.h:56: undefined reference to `__lseek'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/./dl-map-segments.h:60: undefined reference to `__GI___read_nocancel'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/./dl-map-segments.h:73: undefined reference to `rtld_errno'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/librtld.os: in function `_dl_map_object_from_fd':
/home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/dl-load.c:995: undefined reference to `__GI___close_nocancel'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/dl-load.c:1198: undefined reference to `__GI___close_nocancel'
/home/build/black-parrot/external/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/build/black-parrot/external/riscv-gnu-toolchain/build-glibc-linux-rv64imafd-lp64/elf/librtld.os: in function `elf_get_dynamic_info':
/home/build/black-parrot/external/riscv-gnu-toolchain/riscv-glibc/elf/get-dynamic-info.h:48: undefined reference to `rtld_errno'
[... cut ...]
`/home/build/black-parrot/external/Makefile.tools:82: recipe for target '/home/build/black-parrot/external/touchfiles/gnu.70063d751135fb22e754ff14bc969ffe9e905fbd' failed`
`make[2]: *** [/home/build/black-parrot/external/touchfiles/gnu.70063d751135fb22e754ff14bc969ffe9e905fbd] Error 2`
`make[2]: Leaving directory '/home/build/black-parrot'`
`Makefile:62: recipe for target 'tools' failed`
`make[1]: *** [tools] Error 2`
`make[1]: Leaving directory '/home/build/black-parrot'`
`Makefile:24: recipe for target 'prep' failed`
`make: *** [prep] Error 2`
`build@a5569f5593e2:~/black-parrot$ `

gabaxter, 2021-02-12:
I did `virtualenv` apt install and reran `make prep -j 8`.  

This seemed to now finish without error (last 2 lines):
`make[2]: Leaving directory '/home/glenn/work/black-parrot/bp_me/src/asm'`
`make[1]: Leaving directory '/home/glenn/work/black-parrot'`

**So I tried doing:**
`make build_cov.sc sim.sc SUITE=riscv_tests PROG=rsort`

**Which returned** 
`make: *** No rule to make target 'build_cov.sc'.  Stop.`

This might be operator error at this point?  Would not surprise me!  Is there some kind of script to run to set the env up before that works?  I did not see that in readme, but well could have missed it.

dpetrisko, 2021-02-12:
Hi, thanks for the reports.  Regarding virtualenv, it's been added in the prereqs for Ubuntu in dev. Just hasn't percolated to master yet.  It's needed for bsg_sv2v, but nothing else if you need to comment out that build from the Makefile.

https://github.com/black-parrot/black-parrot/blob/dev/docs/getting_started.md#ubuntu

dpetrisko, 2021-02-12:
>This might be operator error at this point? Would not surprise me! Is there some kind of script to run to set the env up before that works? I did not see that in readme, but well could have missed it.

Are you running from bp_top/syn ?

gabaxter, 2021-02-12:
As I said operator error....

It built a bunch of g++ files and then 

The correction for the main readme seems to be prerequisite:
apt-get install -y apt-utils tzdata git vim gettext-base \
        bash autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev gawk \
        build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev \
        wget byacc device-tree-compiler python gtkwave python-yaml pkg-config swig python3-dev \
        python-yaml virtualenv

closed 2021-02-12 by gabaxter

dpetrisko, 2021-02-12:
Thanks, glad it's working!

gabaxter, 2021-02-12:
Thank you guys for all the very quick help.  I hope that my minors setup issues help others avoid these same issues.

gabaxter, 2021-02-16:
Hi Dan,

Can you provide any additional guidance on getting a quad core machine to work through the FPGA flow?  I’ve not been able to find why it does not work for me.  So you know here are the specific repo’s I am currently using:

glenn@xxxx:~/work/black-parrot((HEAD detached at 82e69828))$
glenn@xxxx:~/work/black-parrot/bsg_fpga/vu37p((HEAD detached at 9a1985c))$

Thoughts?

Cheers,

Glenn

dpetrisko, 2021-02-16:
Hi Glenn,

https://github.com/black-parrot-examples/bsg_fpga/blob/master/vu37p/v/design_1_wrapper.v#L27
I modified this line to e_bp_multicore_4_cfg and it synthesized a quad core BP for me. Can you double check that this isn't working for you? Maybe clean the directory to make sure there are no cached build files