black-parrot/black-parrot #777: Regarding Python in Getting Started (Full)

shivampotdar, 2021-02-24:
I was trying to set up the repository using [Full Guide](https://github.com/black-parrot/black-parrot/blob/master/docs/getting_started.md) on Ubuntu 20.04.

I faced two errors under the bsg_sv2v part of the setup. While it might just be me, I think putting up a note here might help someone else.

The errors were - 
1) `pytest-pythonpath` fails to install -
```
black-parrot/external/bsg_sv2v/tools/pypy3/lib-python/3/distutils/cmd.py", line 103, in __getattr__
        raise AttributeError(attr)
    AttributeError: install_layout
```
The fix for this was to add a line - `$(PIP) install --upgrade setuptools` in `$(VIRTUALENV_BUILD_DIR): $(PYPY3_BUILD_DIR)`:146

2) Cannot import main from virtualenv
- I had `virtualenv` under `/home/user/.local/bin`, moved it to a different filename so that `/usr/bin` is used
- `usr/bin/python` pointed to `/etc/alternatives/python` (which again pointed to `/usr/bin/python3`), replaced that with `ln -s /usr/bin/python2 /usr/bin/python` (this part might be optional in some systems)

dpetrisko, 2021-02-25:
Thanks for the report!

developandplay, 2021-03-20:
Yep I also ran into the first issue and adding in that line `$(PIP) install --upgrade setuptools` helped.
The second issue you mentioned probably comes down to how your `virtualenv` is set up so I didn't encounter that.
Additionally my pretty clean ubuntu install was missing `cmake`.

shivampotdar, 2021-03-22:
Thanks for your comments @developandplay. 
I would agree that the 2nd point might be specific to my system. The PATH was polluted as these executables were both in ~/.local and /usr/bin

taylor-bsg, 2021-03-22:
@stdavids see bug report for bsg_sv2v

muwyse, 2021-04-02:
@taylor-bsg @stdavids I ran into a similar problem running `make prep` on dev (c0acd493) when building on Ubuntu 20.04 LTS running on WSL2 (Windows Subsystem for Linux v2). Fix as reported by @WasabiFan was the following:

`cd tools/bsg_sv2v`
`source tools/virtual_env/bin/activate`
`python -m pip install -U setuptools`

dpetrisko, 2021-04-02:
Should that be a PR for here or bsg_sv2v?

Aside: I think we should commit to WSL+Ubuntu as a platform rather than Docker for Windows. 
 Mac I'm not sure, possible Docker is the best thing there...

WasabiFan, 2021-04-02:
Looks like this commit made the change already: https://github.com/bespoke-silicon-group/bsg_sv2v/commit/0b6282083aeca4b5d2e03ebbcec377cd22502952

Bumping the bsg_sv2v submodule in this repo should be sufficient.

muwyse, 2021-04-03:
Awesome, thanks for pointing that out @WasabiFan! I am working on some minor build fixes / fine-tuning so I will incorporate a bump of bsg_sv2v that hopefully fixes the issue.

muwyse, 2021-04-03:
@dpetrisko correct, shouldn't really be a fix in BP repo aside from bumping submodule

muwyse, 2021-04-13:
#836 should address outstanding issues here by bumping bsg_sv2v

muwyse, 2021-04-29:
This should be fixed on dev. Closing issue.

closed 2021-04-29 by muwyse