black-parrot/black-parrot #780: Publish toolchain binaries

dpetrisko, 2021-02-26:
Currently, people need to download BP and their first experience is compiling riscv-gnu-toolchain which takes hours and causes problems. Instead, we should break out our toolchain into a separate repo and then publish binaries using the github release mechanism. This whole process can be automated for minimal effort

dpetrisko, 2021-03-26:
Should be done in: https://github.com/black-parrot-sdk/black-parrot-sdk

developandplay, 2021-04-10:
This would be great in terms of usability.
I created a quick prototype which is not fully tested yet.

Let me know if you have some good ideas on how to integrate the binaries in the existing Makefile.
Currently I'm just extracting the binaries to the places where the make process would put them.
This is the prototype Makefile I came up with:
https://github.com/black-parrot-sdk/black-parrot-sdk/pull/7/files#diff-76ed074a9305c04054cdebb9e9aad2d818052b07091de1f20cad0bbac34ffb52R14-R37

dpetrisko, 2021-09-19:
Fixed in sdk repo

closed 2021-09-19 by dpetrisko