black-parrot/black-parrot #781: UC read/write ops in mem_cmd/mem_resp are unchecked

dpetrisko, 2021-02-26:
https://github.com/black-parrot/black-parrot/blob/dev/bp_common/src/include/bp_common_bedrock_pkgdef.svh#L45

Do we envision using these? Currently, the L2 does not support uncached / passthrough requests so all requests that end up there are cached regardless. We could work around it by bypassing the L2 entirely...

taylor-bsg, 2021-02-26:
Seems like we should outline the usage scenarios (what is the motivation
for pass thru) and also the role of cache management instructions. So for
example, if BP is sharing an external HBM bank with an non-coherent
accelerator, then the one option is to bypass the L2 and have no caching,
the other would be to use the L2 and do software managed coherence. This
case actually applies to BigBlade.

RKNAGA18, 2026-04-01:
thanks @dpetrisko  I’ll add the $error checks for uncached operations in the L2 cache controller and open a PR shortly.