black-parrot/black-parrot #789: Assertions not working in Verilator

dpetrisko, 2021-03-05:
To reproduce:

make build.sc sim.sc COSIM_P=1 SUITE=riscv_tests PROG=rv64ua-p-lrsc CFG=e_bp_unicore_l2_atomic_cfg

This will overflow the watchdog timer but not trigger the assertion. Possibly a verilator bug, it seems to hang at test_bp.cpp L50, does not print enabling asssertions

developandplay, 2021-03-26:
In which directory are you executing this command?
Just asking so that I can do a before and after test with FST on this one.

dpetrisko, 2021-03-26:
bp_top/syn

dpetrisko, 2021-03-26:
Do you think FST will affect assertions?

developandplay, 2021-03-26:
No I would assume the bug remains.
Nevertheless I still wanted to check since with FST the reset is defined in verilog instead of systemc.

developandplay, 2021-03-26:
I was running this command in bp_top/syn but was missing the riscv_tests.
Then I ran `make -C sdk prog` to manually generate them so it's all fine but I was just wondering if this is by design.

Also for some reason after running that command the tests were in `black-parrot/sdk/prog/riscv-tests` but for
```
make build.sc sim.sc COSIM_P=1 SUITE=riscv_tests PROG=rv64ua-p-lrsc CFG=e_bp_unicore_l2_atomic_cfg
```
they are supposed to be in `black-parrot/sdk/prog/riscv_tests`.

developandplay, 2021-03-26:
And yes unfortunately nothing changed...
I'm now getting the output before FST:
```
Lowering reset
Running with Dromajo cosimulation
[DEBUG] DUT raised exception 2
csr_read: invalid CSR=0x3b0
[error] HARTID: 0
[error] EMU PC 00000000800001dc, DUT PC 00000000800001dc
[error] EMU INSN 00052583, DUT INSN 00052583
[error] EMU WDATA 0000000000000001, DUT WDATA 0000000000000000
[error] EMU MSTATUS a00000080, DUT MSTATUS a00000080
[error] DUT pending exception -1 pending interrupt -1
NBF loader done!
COSIM_FAIL
- /home/martin/coding/black-parrot/bp_top/test/common/bp_nonsynth_cosim.sv:222: Verilog $finish
Finishing test
Executing final
[CORE0 STATS]
        clk   :                  353
        instr :                   84
        mIPC  :                  237
Exiting
```
and after FST:
```
\______   \\_   _____/ /   _____/\_   _____/\__    ___/
 |       _/ |    __)_  \_____  \  |    __)_   |    |   
 |    |   \ |        \ /        \ |        \  |    |  1->0 time =      12500
 |____|_  //_______  //_______  //_______  /  |____|   
 ASYNC  \/         \/         \/         \/            
__________ ___________  _______________________________
\______   \\_   _____/ /   _____/\_   _____/\__    ___/
 |       _/ |    __)_  \_____  \  |    __)_   |    |   
 |    |   \ |        \ /        \ |        \  |    |  1->0 time =     100000
 |____|_  //_______  //_______  //_______  /  |____|   
 ASYNC  \/         \/         \/         \/            
Running with Dromajo cosimulation
[DEBUG] DUT raised exception 2
csr_read: invalid CSR=0x3b0
[error] HARTID: 0
[error] EMU PC 00000000800001dc, DUT PC 00000000800001dc
[error] EMU INSN 00052583, DUT INSN 00052583
[error] EMU WDATA 0000000000000001, DUT WDATA 0000000000000000
[error] EMU MSTATUS a00000080, DUT MSTATUS a00000080
[error] DUT pending exception -1 pending interrupt -1
NBF loader done!
COSIM_FAIL
- /home/martin/coding/black-parrot/bp_top/test/common/bp_nonsynth_cosim.sv:222: Verilog $finish
Finishing test
Executing final
[CORE0 STATS]
        clk   :                  353
        instr :                   84
        mIPC  :                  237
Exiting
```

dpetrisko, 2021-03-26:
Yeah the repo was renamed riscv-tests after that issue was written

dpetrisko, 2021-03-26:
COSIM_P=0 should trigger the bug

developandplay, 2021-03-26:
Wait so the output I got with COSIM_P=1 is the expected output then?
I'm not very fimiliar with dromajo so I didn't know how to interpret the exception...
If I run it with COSIM_P=0 then it's haging on the reset:
```
## bsg_mem_1rw_sync_mask_write_byte: instantiating data_width_p=        64, els_p=       512 (test_bp.testbench.wrapper.unicore.dut.core_minimal.be.calculator.pipe_mem.dcache.d[6].data_mem)
## bsg_mem_1rw_sync_mask_write_byte: instantiating data_width_p=        64, els_p=       512 (test_bp.testbench.wrapper.unicore.dut.core_minimal.be.calculator.pipe_mem.dcache.d[7].data_mem)
Lowering reset
```
If I run it with COSIM_P=0 and after enabling FST then the resets won't even get triggered:
```
## bsg_mem_1rw_sync_mask_write_byte: instantiating data_width_p=        64, els_p=       512 (testbench.testbench.wrapper.unicore.dut.core_minimal.be.calculator.pipe_mem.dcache.d[5].data_mem)
## bsg_mem_1rw_sync_mask_write_byte: instantiating data_width_p=        64, els_p=  
```
It seems like it is already hanging during initialization.

dpetrisko, 2021-03-26:
This issue is really about assertions, so just add an
```initial assert (false) else $finish()``` to some module. The assertions don't get triggered

developandplay, 2023-05-03:
@dpetrisko Random one and you might have figured this out long ago.
I was recently playing around with assertions in Verilator for a testbench.
After some trial and error I realised that I needed to enable assertions when invoking verilator with `--assert`.
You should probably double check if this leads to some slower simulation speeds but it might be handy so here is the commit.

dpetrisko, 2023-05-03:
Nice!

dpetrisko, 2023-07-18:
Closing; Added to makefiles

closed 2023-07-18 by dpetrisko