black-parrot/black-parrot #792: Add assertion for dram size

dpetrisko, 2021-03-05:
2**32 <= Max DRAM size < paddr_width_p

RKNAGA18, 2026-03-12:
Hi @dpetrisko,
I'm currently looking into implementing this assertion in testbench.sv.

The upper bound if (daddr_width_p >= paddr_width_p) is currently implemented, but the lower bound is missing. I was about to add if (daddr_width_p < 32) as requested, but while testing, I noticed that the default testbench (e_bp_default_cfg) actually runs with a daddr_width_p smaller than 32 bits.

If I add the < 32 fatal assertion, it will immediately break the default testbench. Should we update the default configurations to have a minimum 4GB memory, or should this assertion only be applied in specific configs?

dpetrisko, 2026-03-12:
The constraint is driven by L1 cacheable address space (caddr) being 32b as the local I/O space is allocated to 0x0000_0000 to 0x07FF_FFFF. The dram address space (daddr) must be larger so is minimally 33b. 

Probably the 32b could be lowered, but most RISCV software is written with 0x8000_0000 as the target dram base so there's some motivation to keep that and prevent aliasing. I'm not surprised the testbench still works as it's not exercising the entire I/O space. 

So we would want the assertion to be the tested range of caddr_width >= 32 and daddr_width > caddr_width