black-parrot/black-parrot #798: Enable FST generation in Verilator

dpetrisko, 2021-03-14:
GTKWave works better / faster with FST files.  It should be pretty easy to enable this flow, with substantial runtime benefits

developandplay, 2021-03-22:
I looked into this issue to familiarize myself with the workflow of the project for GSoC.
Now I did manage to enable `--trace-fst` but after running into a few linking issues I realized it is incompatible with Systemc.
From the manual:
> How do I generate FST waveforms (aka dumps or traces) in SystemC?
> The FST library from GTKWave does not currently support SystemC; use VCD format instead.

Using the `CONNECTING TO C ++` section of the manual I was able to add the undefined `sc_time_stamp()`. However now I'm stuck on the remaining `undefined reference to 'main'`. Is there a simple way to switch from Systemc to C++ or is this a more involved task?

dpetrisko, 2021-03-22:
> I realized it is incompatible with Systemc.

Yep.

> Is there a simple way to switch from Systemc to C++ or is this a more involved task?

Our test_bp.cpp is pretty close to the example SystemC/C++ testbench Verilator provides. The most notable change is that we need to manage both the core clock and DRAM clock. I think without the SystemC clock class, this involves creating two clock threads. Nothing too crazy, but non-trivial

taylor-bsg, 2021-03-22:
Not sure it helpful or different but Dustin has the dpi based clock
generator classes.

dpetrisko, 2021-03-22:
Oh yeah that might be useful:

https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_test/bsg_nonsynth_dpi_clock_gen.cpp

Using this would mean that test_bp would have an ifdef VERILATOR selecting between the clock gens, and then test_bp.cpp would be instantiating these dpi clocks

developandplay, 2021-03-23:
Ok so I made some progress but I'm stuck on the dpi methods. Unfortunately they are resulting in undefined reference erros.
Now I did find the [testing/bsg_test/bsg_nonsynth_dpi/](https://github.com/bespoke-silicon-group/basejump_stl/blob/7556378810412be47e118b08910c02883936a64c/testing/bsg_test/bsg_nonsynth_dpi/) as a reference. This helped me fix quite a few prior errors but I'm not sure which flag/command ensures that dpi methods are included.

dpetrisko, 2021-03-23:
Seems it's defined there:

https://github.com/bespoke-silicon-group/basejump_stl/blob/c63085a0af9e0b72f7ac42099aca9d9eb4c85628/bsg_test/bsg_nonsynth_dpi_clock_gen.v

Have you added this to the test flist?

developandplay, 2021-03-23:
Yep I also added `bsg_defines.v` as `bsg_nonsynth_dpi_clock_gen.v` seems to depend on it.
I now put up a commit of the current state that leads to the error.
https://github.com/black-parrot/black-parrot/commit/b7dd19d29362de6f5e08c1315c304f528ce8bd1d

Just running it with the fst flag enabled and without including anything `bsg_nonsynth_dpi_clock_gen.hpp` related works fine. Of course that wouldn't the correct simulation.

dpetrisko, 2021-03-23:
Cool!

test_bp.sv is only compiled for VCS testbenches currently, so it's not finding the verilog clock_gen instantiations.  You'll need to add test_bp.sv to the flist in the Makefile.verilator as shown here:
[testbench Makefile.verilator](https://github.com/black-parrot/black-parrot/blob/master/bp_top/test/tb/bp_tethered/Makefile.verilator#L9). test_bp.sv should have `ifdef VERILATOR guards so that VCS uses the regular clock generators and Verilator uses the dpi clock gen.

Other feedback:

- Add all sources and includes to: [testbench flist.vcs](https://github.com/black-parrot/black-parrot/blob/master/bp_top/test/tb/bp_tethered/flist.vcs) rather than in the Makefile
- Don't add nonsynthesizable sources to the [synth flist.vcs](https://github.com/black-parrot/black-parrot/blob/master/bp_top/syn/flist.vcs)
- You can then get rid of get_sim_period and get_dram_period DPI calls in the testbench

developandplay, 2021-03-24:
So with these changes `test_pb.sv` becomes the top module for both vcs and verilator right?
Obviously after making it the top module now the paramaters for `testbench.sv` are not in the design. However when using vcs that's also the case and it doesn't lead to issues. So the configuration must be a bit different in that regard.
```
%Error: Parameters from the command line were not found in the design: amo_en_p br_profile_p cce_trace_p checkpoint_p cmt_trace_p core_profile_p cosim_instr_p cosim_memsize_p cosim_p dcache_trace_p dram_trace_p icache_trace_p lce_trace_p no_bind_p pc_profile_p preload_mem_p vm_trace_p warmup_instr_p
%Error: Exiting due to 1 error(s)
```

Also do you want to switch to fst completly or make it an option?

dpetrisko, 2021-03-24:
>Also do you want to switch to fst completly or make it an option?
Completely

My guess is that VCS pushes down parameters from the toplevel whereas Verilator requires them to be explicitly passed.  At that point, there's no real reason for test_bp to exist, it's going to be a maintenance overhead. I would just put the clock+reset generators in the testbench and have testbench.sv as the toplevel for VCS (and the Verilated toplevel for Verilator). This would get rid of test_bp.sv

developandplay, 2021-03-24:
Great now I finally got everything to build.
The only thing missing now is figuring out how to deal with this lines in verilator so that the behavior is also as expected.
https://github.com/black-parrot/black-parrot/blob/74fb1732207b63be4c22b9cc9ab52e2a2b36fc2e/bp_top/test/tb/bp_tethered/test_bp.sv#L57-L63

dpetrisko, 2021-03-24:
You can just ifndef VERILATOR that -- test_bp.cpp handles the assertion enable. (I'm not actually sure, does Verilator support $asserton/off?)

developandplay, 2021-03-24:
Ok great I wasn't sure if both asserts are necessary.
No verilator does not seem to support $asserton/off so the entire block seems pointless.

If I understand the code correctly then manually assigning reset in `test_bp.cpp` also should not be necessary anymore.
I'm talking about these lines:
https://github.com/black-parrot/black-parrot/blob/74fb1732207b63be4c22b9cc9ab52e2a2b36fc2e/bp_top/test/tb/bp_tethered/test_bp.sv#L24-L32

At this point I'm getting this output:
```
__________ ___________  _______________________________
\______   \\_   _____/ /   _____/\_   _____/\__    ___/
 |       _/ |    __)_  \_____  \  |    __)_   |    |   
 |    |   \ |        \ /        \ |        \  |    |  0->1 time =          0
 |____|_  //_______  //_______  //_______  /  |____|   
 ASYNC  \/         \/         \/         \/            
__________ ___________  _______________________________
\______   \\_   _____/ /   _____/\_   _____/\__    ___/
 |       _/ |    __)_  \_____  \  |    __)_   |    |   
 |    |   \ |        \ /        \ |        \  |    |  0->1 time =          0
 |____|_  //_______  //_______  //_______  /  |____|   
 ASYNC  \/         \/         \/         \/            
Opening dump file
Raising reset
Lowering reset
```
Which shows the two first resets were triggered and set to 1 at time 0 like they should be.
However the seconds reset after 20 cycles are not triggered and `Verilated::gotFinish()` won't turn true either.
Maybe I'm missing something when it comes to letting the clock run.

This is the current state: a7a6bb81b4c7360519ff6223640e075f8cdd07d6
The commit where I merged testbench.v and test_bp.v is this one: 23233527ff11ea6791ed9cfa364813fdfb97ef06

dpetrisko, 2021-03-24:
In test_bp.cpp, rather than waiting for 20 cycles, you should poll the reset signal of the testbench in test_bp.cpp. That will let you know when to turn on assertions. You can then get rid of the lowering/raising reset prints in the test_bp.cpp.

When you say the ram clock doesn't work, what do you mean? Is it printing out the initial block shown here for both clocks? 

```
   initial begin
      $display("BSG INFO: bsg_nonsynth_dpi_clock_gen (initial begin)");
      $display("BSG INFO:     Instantiation: %M");
      $display("BSG INFO:     cycle_time_p  = %d", cycle_time_p);
      hierarchy = $sformatf("%m");
      id = bsg_dpi_clock_gen_register(cycle_time_lp, hierarchy);
   end
```

developandplay, 2021-03-24:
Oh true. Right now the `test_bp.cpp` doesn't know when to turn assertions on/off.

I just meant if I use a debug statement that gives me the current cycle for both clocks like this:
```
  int cycle = 0;
  int dramcycle = 0;

  always @(posedge clk_i) begin
    cycle <= cycle +1;
    $display("BSG DBGINFO: top -- cycle %d", cycle);
    $display("BSG DBGINFO: top -- dramCycle %d", dramcycle);
    $display("BSG DBGINFO: top -- reset %d", reset_i);
    $display("BSG DBGINFO: top -- dramReset %d", dram_reset_i);
  end

  always @(posedge dram_clk_i) begin
    dramcycle <= dramcycle +1;
  end
```
Then the `cycles` will update but the `dramCycles` will stay at 1. Or is that expected behaviour?
```
BSG DBGINFO: top -- cycle 276891
BSG DBGINFO: top -- dramCycle 1
BSG DBGINFO: top -- reset 0
BSG DBGINFO: top -- dramReset 1
```

Also yes it's printing out both clocks during initialization. So that works as expected:
```
BSG INFO: bsg_nonsynth_dpi_clock_gen (initial begin)
BSG INFO:     Instantiation: testbench.testbench.clock_gen
BSG INFO:     cycle_time_p  =                 5000
BSG INFO: bsg_nonsynth_dpi_clock_gen (initial begin)
BSG INFO:     Instantiation: testbench.testbench.dram_clock_gen
BSG INFO:     cycle_time_p  =                 1250
```

dpetrisko, 2021-03-25:
Congrats! We think you've found a bug in https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_test/bsg_nonsynth_dpi_clock_gen.cpp.  It's never been tested with multiple clock generators. If you want to delve deep into DPI feel free to take a look. I raised an issue for us on BaseJumpSTL.

Thanks, this is great!

drichmond, 2021-03-25:
Can you try this PR?

https://github.com/bespoke-silicon-group/basejump_stl/pull/422

developandplay, 2021-03-25:
Wow that was quick!
Thank you @drichmond. It's working now! :partying_face:
@dpetrisko Now that I have the demo running I'll create a PR with some more polished code.
This is a great task to get to know the entire build process and get some first exposure to the basejump modules.

dpetrisko, 2021-03-25:
Looking forward to it. When you make the PR, could you run a couple test programs w/wo your change and verify the file size reduction? I assume you have the sdk working, maybe beebs/aha-compress riscv-tests/towers and bp-tests/hello_world?

developandplay, 2021-03-26:
So all of the tests are running successfully for top.
For example for riscv-test/towers:

- dump.vcd is 573.2mb
- dump.fst is 9.6mb

developandplay, 2021-03-26:
Now I have a few questions:

- The test benches for fe, be and me are still using vcd...Which tests can I run on them after switching to fst?
- It seems like it is a convention that .v and .cpp of the testbench have the same name...Should I rename the test_bp.cpp to testbench.cpp?
- The commands currently end with .sc e.g. build.sc...Was this this because they were using the -sc flag previously? If so should rename them to .cc?

dpetrisko, 2021-03-26:
>dump.vcd is 573.2mb
dump.fst is 9.6mb
>I'm not running the vcs tests yet...Is there a public/free version of vcs?

No, usually people get a license through their university or company. I just ran towers and got 6.0mb, so it's better but fst is 90% as good. Impressive!

>The test benches for fe, be and me are still using vcd...Which tests can I run on them after switching to fst?

They have their own testlists in the testbench folder. If you just run make build.sc sim.sc in their respective bp\_x/syn, it'll run a test so you can validate the change. Not too concerned with the compression ratio for those, just that they work.

>The commands currently end with .sc e.g. build.sc...Was this this because they were using the -sc flag previously? If so should rename them to .cc?

Keep it the same for now

taylor-bsg, 2021-03-26:
Great progress with the .fst stuff!

M

developandplay, 2021-03-27:
@dpetrisko I'm a bit confused about the **fe** and **be** testbenches.
Both of them say that verilator is not supported but they still have `test_bp.cpp` and `test_bp.sv`. Could we delete those or is there some dependency I'm missing?

For the **me** testbench I'm getting this output:
```
%Warning-MODDUP: /home/martin/coding/black-parrot/external/basejump_stl/bsg_dataflow/bsg_round_robin_n_to_1.v:18:8: Duplicate declaration of module: 'bsg_round_robin_n_to_1'
   18 | module bsg_round_robin_n_to_1 #(parameter width_p = -1
      |        ^~~~~~~~~~~~~~~~~~~~~~
                 /home/martin/coding/black-parrot/external/basejump_stl/bsg_dataflow/bsg_round_robin_n_to_1.v:18:8: ... Location of original declaration
   18 | module bsg_round_robin_n_to_1 #(parameter width_p = -1
      |        ^~~~~~~~~~~~~~~~~~~~~~
                 ... Use "/* verilator lint_off MODDUP */" and lint_on around source to disable this message.
%Error: /home/martin/coding/black-parrot/bp_me/test/common/bp_cce_mmio_cfg_loader.sv:227:25: Can't find definition of variable: 'cfg_reg_reset_gp'
                                                                                           : ... Suggested alternative: 'cfg_reg_freeze_gp'
  227 |           cfg_addr_lo = cfg_reg_reset_gp;
      |                         ^~~~~~~~~~~~~~~~
%Error: testbench.sv:29:46: Can't find definition of variable: 'BP_DRAM_FLOWVAR'
   29 |    , parameter dram_type_p                 = BP_DRAM_FLOWVAR  
      |                                              ^~~~~~~~~~~~~~~
%Error: Exiting due to 2 error(s)
``` 
No matter whether I'm using FST or VCD. Is there a branch that has these issues fixed?

dpetrisko, 2021-03-27:
Sorry bout that, rebase on from latest dev for the fix. (or apply this patch)

diff --git a/bp_common/src/include/bp_common_cfg_bus_pkgdef.svh b/bp_common/src/include/bp_common_cfg_bus_pkgdef.svh
index 5c12395b2..e1b49f7b4 100644
--- a/bp_common/src/include/bp_common_cfg_bus_pkgdef.svh
+++ b/bp_common/src/include/bp_common_cfg_bus_pkgdef.svh
@@ -22,7 +22,6 @@
     ,e_cce_mode_normal  = 1
   } bp_cce_mode_e;

```
-  // TODO: This is out of date.  The actual map shouldn't matter much, but we should decide...
   // The overall memory map of the config link is:
   //   16'h0000 - 16'h001f: chip level config
   //   16'h0020 - 16'h003f: fe config
@@ -35,6 +34,7 @@
   localparam cfg_data_width_gp = 64;

   localparam cfg_base_addr_gp          = 'h0200_0000;
+  localparam cfg_reg_reset_gp          = 'h0001; // Unused
   localparam cfg_reg_freeze_gp         = 'h0002;
   localparam cfg_reg_core_id_gp        = 'h0005;
   localparam cfg_reg_did_gp            = 'h0006;
```

dpetrisko, 2021-03-27:
>Both of them say that verilator is not supported but they still have test_bp.cpp and test_bp.sv. Could we delete those or is there some dependency I'm missing?

Both the FE and BE testbenches require "preloading" the memory because they don't currently have NBF loaders in the testbench. It's something we should fix at some point, but for now only VCS supports this preloading. I'd still like the testbenches to be the same as the top/me just for consistency when we do update

developandplay, 2021-03-28:
Thanks for the fix! That helped.
Now I'm runnning into errors related to test programs being in the wrong location.
For example gcc is complaining about not finding `black-parrot/bp_me/src/asm/roms/mesi.mem` I guess this might related to the restructuring.

developandplay, 2021-03-28:
This patch did the trick for me:
```
diff --git a/bp_me/test/tb/bp_cce/Makefile.params b/bp_me/test/tb/bp_cce/Makefile.params
index cf744e08..e96b8def 100644
--- a/bp_me/test/tb/bp_cce/Makefile.params
+++ b/bp_me/test/tb/bp_cce/Makefile.params
@@ -13,7 +13,7 @@ export SKIP_INIT_P ?= 0
 export LCE_MODE_P ?= 0
 
 COH_PROTO   ?= mesi
-CCE_MEM_PATH = $(BP_ME_DIR)/src/asm/roms
+CCE_MEM_PATH = $(BP_SDK_DIR)/bedrock/roms
 CCE_MEM      = $(COH_PROTO).mem
 
 NUMS = $(shell seq 0 `expr $(NUM_LCE_P) - 1`)
```

Though maybe there is more to look out for. I'm not too familiar with the memory part yet.

dpetrisko, 2021-03-28:
Yeah that looks right

dpetrisko, 2021-03-30:
Closing from #820

closed 2021-03-30 by dpetrisko