black-parrot/black-parrot #831: Dromajo build fails when running 'make prep'

kf7lsu, 2021-04-09:
Hi,

I'm working on getting set up with Black Parrot on Ubuntu 18.04. I have all of the required packages listed in "Getting Started (Full)". I am attempting to set this up as root, as a way to eliminate any suspicion of permissions errors. When I run 'make prep', it is able to run up until it attempts to build Dromajo. When it attempts to build Dromajo, it checks for CMakeLists.txt in a build folder, I've copied the relevant portion of the build log below. 

I am guessing that either something is going wrong recursively or I am missing an unlisted prerequisite.

Any help troubleshooting this problem would be much appreciated.

`cd /home/mtrahms/EE470SP21/black-parrot/sdk; git submodule update --init --checkout --depth=1
Submodule path 'bedrock': checked out '37b63d6435037814446692f8d5f20727aebab2e2'
Submodule path 'beebs': checked out 'ebe893d57167f170793d067c2d4d97e5adfe5889'
Submodule path 'bootrom': checked out 'fbfb7341f699646de0d9feed225474023b762a33'
Submodule path 'bp-demos': checked out 'f075c088775fb691d766b15e03c8282eb3dca7b9'
Submodule path 'bp-tests': checked out '889dacb66b0ae362cb79e1a327415b1d1548364a'
Submodule path 'coremark': checked out '69b6914aea64c461277f476195a314080db170ac'
Submodule path 'dromajo': checked out '4cbf5d91b880e172f15e0658cc72d4d6426ddcaa'
Submodule path 'linux': checked out 'fca605b198372ecb71525c05d727097076c7dcb9'
Submodule path 'opensbi': checked out 'eac24d8778eb8497629431af523d05fd54b42d42'
Submodule path 'perch': checked out 'db209a396bfed0db542463d3f348d82b34e5ef13'
Submodule path 'riscv-dv': checked out '8ec8dfcb7fb9902315878688972749962c9c3475'
Submodule path 'riscv-gnu-toolchain': checked out 'b2924c5d26c1b9a5628bc51acdf28b7045a6542f'
Submodule path 'riscv-opcodes': checked out '8ca982845164f4995003eb29d11d7ff8bd5bd17e'
Submodule path 'riscv-tests': checked out '82ad412f97785928d2add4125d7391a26cb08437'
Submodule path 'spec2000': checked out '30f709e3c2d736257c7510dee603b0893427e528'
make dromajo
make[2]: Entering directory '/home/mtrahms/EE470SP21/black-parrot/sdk'
rm -rf /home/mtrahms/EE470SP21/black-parrot/sdk/install/touchfiles/dromajo.*
cd /home/mtrahms/EE470SP21/black-parrot/sdk; git submodule update --init  --recursive --checkout /home/mtrahms/EE470SP21/black-parrot/sdk/dromajo
make dromajo_build
make[3]: Entering directory '/home/mtrahms/EE470SP21/black-parrot/sdk'
mkdir -p /home/mtrahms/EE470SP21/black-parrot/sdk/dromajo/build
cmake -S /home/mtrahms/EE470SP21/black-parrot/sdk/dromajo -B /home/mtrahms/EE470SP21/black-parrot/sdk/dromajo/build
CMake Error: The source directory "/home/mtrahms/EE470SP21/black-parrot/sdk/dromajo/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
/home/mtrahms/EE470SP21/black-parrot/sdk/Makefile.tools:26: recipe for target 'dromajo_build' failed
make[3]: *** [dromajo_build] Error 1
make[3]: Leaving directory '/home/mtrahms/EE470SP21/black-parrot/sdk'
/home/mtrahms/EE470SP21/black-parrot/sdk/Makefile.tools:41: recipe for target '/home/mtrahms/EE470SP21/black-parrot/sdk/install/touchfiles/dromajo.4cbf5d91b880e172f15e0658cc72d4d6426ddcaa' failed
make[2]: *** [/home/mtrahms/EE470SP21/black-parrot/sdk/install/touchfiles/dromajo.4cbf5d91b880e172f15e0658cc72d4d6426ddcaa] Error 2
make[2]: Leaving directory '/home/mtrahms/EE470SP21/black-parrot/sdk'
Makefile:19: recipe for target 'sdk_lite' failed
make[1]: *** [sdk_lite] Error 2
make[1]: Leaving directory '/home/mtrahms/EE470SP21/black-parrot/sdk'
Makefile:18: recipe for target 'sdk' failed
make: *** [sdk] Error 2`

dpetrisko, 2021-04-09:
Hi, try updating your cmake

https://github.com/black-parrot/black-parrot/pull/808

dpetrisko, 2021-04-09:
> am attempting to set this up as root

Do not do this

kf7lsu, 2021-04-09:
Thank you for the tips. Updating cmake solved the problem.

closed 2021-04-09 by kf7lsu

taylor-bsg, 2021-04-09:
I suggest adding a check for cmake version would help avoid many similar
error requests.