black-parrot/black-parrot #832: Division by zero results are incorrect according to the specification

sammy17, 2021-04-09:
div instructions (including 32bit, signed, unsigned variants) return value is incorrect according to the specification when the divisor is is zero.
rem instructions (including 32bit, signed, unsigned variants) return value is incorrect according to the specification when either the divisor is zero or dividend is zero. 
**Please refer Volume I: RISC-V Unprivileged ISA V20191213 - Table 7.1**
When executed with dromajo, mismatches from these instructions are reported.

For example, following instructions will cause mismatches with dromajo Vs DUT simulation:
divu s9,s5,zero
rem s1,s8,zero
divuw t5,a5,zero
rem s9,zero,ra

**How to reproduce:**
cd sdk/riscv-dv/
# set RISCV_GCC and RISCV_OBJCOPY envs
python run.py --steps gen,gcc_compile --simulator vcs --iterations 1 -o out/ -ct user_extension -cs user_extension -tn riscv_arithmetic_basic_test --isa rv64gc --mabi lp64 --seed 12345
mkdir -p ../prog/riscv-dv
cp  out/asm_tests/riscv_arithmetic_basic_test_0.riscv ../prog/riscv-dv/
cd ../../bp_top/syn/
make build.v sim.v  SUITE=riscv-dv COSIM_P=1 PROG=riscv_arithmetic_basic_test_0

dpetrisko, 2021-04-09:
What is the specific case that is incorrect?

dpetrisko, 2021-04-09:
A simple C program that demonstrates the bug(s) would be very helpful

sammy17, 2021-04-09:
By specific case you mean specific application in C?
If so I don't have a ready C code that will trigger division by zero right now. 
But I have attached one of the assembly programs that triggered this issue. 
As mentioned, write back value is not according to the specification when there is a division by zero or remainder when divided by zero. (I encountered these bugs at instruction level, not application level).

Let me know if you have any doubts.
[riscv_arithmetic_basic_test_1.zip](https://github.com/black-parrot/black-parrot/files/6283011/riscv_arithmetic_basic_test_1.zip)

dpetrisko, 2021-04-09:
I mean the input, output, expected and actual.

For instance:
x = 3 / 0;
Should be -1, is 231414.

We test the divide by zero case for all of those instructions, so I'm curious for what cases this occurs.
https://github.com/black-parrot-sdk/riscv-tests/blob/82ad412f97785928d2add4125d7391a26cb08437/isa/rv32um/divu.S#L28

What is the output of the dromajo mismatch?

sammy17, 2021-04-09:
Following image has few mismatches we encountered, you can see the dromajo output and DUT output columns. 
[image]

For example, first entry in the table is **divuw a7,t1,zero**. Expected write back value in this case is -1(ffffffff), but DUT write back value is 0.

dpetrisko, 2021-05-06:
An update on this: I found the bug, and it's actually an interesting hazard condition.  On writes to x0, we suppress the write control signal before it goes into the pipeline to avoid arbitration with 0 later in the pipeline. These particular division instructions read x0, which normally never triggers bypass conditions because the write control signal is zero. The bug is that division -> x0, ostensibly a nop, does not suppress the write control signal, it actually writes to zero. Therefore, the write data gets bypassed and we read x0 as something non-zero.

I'm going to produce a minimal test to reproduce this bug and add it to our regression. Thanks, great find!

dpetrisko, 2021-05-06:
https://github.com/black-parrot-sdk/bp-tests/pull/9
#859

dpetrisko, 2021-05-12:
Fixed with #859

PR #859 (2021-05-06): Fix division to zero bug
- Only do the division if we actually want to do it
- Specifically, don't write to zero on division instruction stored to zero

Fixes #832
closed 2021-05-12 by dpetrisko