black-parrot/black-parrot #842: Syntax error in bp_be_dcache.sv

MikePopoloski, 2021-04-18:
A variable declared in this file is referenced prior to its declaration, which is not actually legal according to the LRM. Verilator and VCS aren't great about diagnosing these issues, but other tools that more strictly conform to the LRM issue an error:
```
black-parrot/bp_be/src/v/bp_be_dcache/bp_be_dcache.sv:409:13: error: identifier 'stat_mem_data_lo' used before its declaration
    (.lru_i(stat_mem_data_lo.lru)
            ^~~~~~~~~~~~~~~~
black-parrot/bp_be/src/v/bp_be_dcache/bp_be_dcache.sv:552:25: note: declared here
  bp_dcache_stat_info_s stat_mem_data_lo;
                        ^
```

I think the fix is as easy as moving the declaration up a bit earlier in the file. If you'd like a PR I'd be happy to open one.

dpetrisko, 2021-04-18:
Thanks Mike. I’d prefer to move the LRU encode below the stat_mem (and move lru_encode signal declaration right before usage). Would appreciate a PR!

Can I ask which tool you’re running through?

MikePopoloski, 2021-04-18:
OK, I'll take a crack at it.

The error snippet is actually from my own linting tool that I've built called [slang](https://github.com/MikePopoloski/slang) but one of the commercial tools I have access to gives a similar result. Most tools I've tried this with don't error though, probably because the dotted name causes them to try a hierarchical lookup and then they don't go back and verify the location once they find that it's a member access.

dpetrisko, 2021-04-18:
Very interesting, thanks!

MikePopoloski, 2021-04-18:
Thanks for merging!

closed 2021-04-18 by MikePopoloski