black-parrot/black-parrot #847: bp-tests/uncached_mode_test passes on default configuration but fails with e_bp_unicore_l1_tiny_cfg

WasabiFan, 2021-04-22:
On commit: a79cc8d5bb118940c353bd24c1c511e00ab4599c

The following command passes the test:

```
make build_dump.v sim_dump.v SUITE=bp-tests PROG=uncached_mode_test CMT_TRACE_P=1
```

However, the following fails:

```
make build_dump.v sim_dump.v SUITE=bp-tests PROG=uncached_mode_test CFG=e_bp_unicore_l1_tiny_cfg CMT_TRACE_P=1
```

The difference is only the bp configuration.

This testbench cannot be tested in cosim since Dromajo doesn't understand the particular custom CSR in use in this test.

[tinyparrot_cfg_commit_0.trace.txt](https://github.com/black-parrot/black-parrot/files/6354526/tinyparrot_cfg_commit_0.trace.txt)
[prog.dump.txt](https://github.com/black-parrot/black-parrot/files/6354527/prog.dump.txt)
[default_cfg_commit_0.trace.txt](https://github.com/black-parrot/black-parrot/files/6354528/default_cfg_commit_0.trace.txt)
[trace_diff.txt](https://github.com/black-parrot/black-parrot/files/6354535/trace_diff.txt)

<details>
  <summary>Visualized diff</summary>

```diff
--- default_cfg_commit_0.trace.txt      2021-04-21 16:56:09.290726200 -0700
+++ tinyparrot_cfg_commit_0.trace.txt   2021-04-21 16:57:57.116970500 -0700
@@ -219,137 +219,24 @@
 0 00800002c4 00001797 000000da 0f 00000000800012c4
 0 00800002c8 01478793 000000db 0f 00000000800012d8
 0 00800002cc 0007b783 000000dc 0f 0000000080000000
-0 00800002d0 0007b783 000000dd 0f 0000000000000060
+0 00800002d0 0007b783 000000dd 0f 0000000000000090
 0 00800002d4 fef42623 000000de
-0 00800002d8 fec42783 000000df 0f 0000000000000060
-0 00800002dc 0007871b 000000e0 0e 0000000000000060
+0 00800002d8 fec42783 000000df 0f 0000000000000090
+0 00800002dc 0007871b 000000e0 0e 0000000000000090
 0 00800002e0 06000793 000000e1 0f 0000000000000060
 0 00800002e4 00f70663 000000e2
-0 00800002f0 00001797 000000e3 0f 00000000800012f0
-0 00800002f4 ff078793 000000e4 0f 00000000800012e0
-0 00800002f8 0007b783 000000e5 0f 0000000080001060
-0 00800002fc 0007b783 000000e6 0f 0000000000000040
-0 0080000300 fef42623 000000e7
-0 0080000304 00001797 000000e8 0f 0000000080001304
-0 0080000308 fcc78793 000000e9 0f 00000000800012d0
-0 008000030c 0007b783 000000ea 0f 0000000000200043
-0 0080000310 00100713 000000eb 0e 0000000000000001
-0 0080000314 00e7b023 000000ec
-0 0080000318 00001797 000000ed 0f 0000000080001318
-0 008000031c fc078793 000000ee 0f 00000000800012d8
-0 0080000320 0007b783 000000ef 0f 0000000080000000
-0 0080000324 05000713 000000f0 0e 0000000000000050
-0 0080000328 00e7b023 000000f1
-0 008000032c 00001797 000000f2 0f 000000008000132c
-0 0080000330 ff478793 000000f3 0f 0000000080001320
-0 0080000334 0007b783 000000f4 0f 0000000080009060
-0 0080000338 04000713 000000f5 0e 0000000000000040
-0 008000033c 00e7b023 000000f6
-0 0080000340 81018693 000000f7 0d 00000000800012e0
-0 0080000344 0006b683 000000f8 0d 0000000080001060
-0 0080000348 00001797 000000f9 0f 0000000080001348
-0 008000034c fe078793 000000fa 0f 0000000080001328
-0 0080000350 0007b783 000000fb 0f 0000000000000040
-0 0080000354 05000713 000000fc 0e 0000000000000050
-0 0080000358 00e7b023 000000fd
-0 008000035c 0006b703 000000fe 0e 0000000000000040
-0 0080000360 fec42783 000000ff 0f 0000000000000040
-0 0080000364 0007871b 00000100 0e 0000000000000040
-0 0080000368 04000793 00000101 0f 0000000000000040
-0 008000036c 00f70663 00000102
-0 0080000378 00001797 00000103 0f 0000000080001378
-0 008000037c f6078793 00000104 0f 00000000800012d8
-0 0080000380 0007b783 00000105 0f 0000000080000000
-0 0080000384 0007b783 00000106 0f 0000000000000050
-0 0080000388 fef42623 00000107
-0 008000038c fec42783 00000108 0f 0000000000000050
-0 0080000390 0007871b 00000109 0e 0000000000000050
-0 0080000394 05000793 0000010a 0f 0000000000000050
-0 0080000398 00f70663 0000010b
-0 00800003a4 00001797 0000010c 0f 00000000800013a4
-0 00800003a8 f4478793 0000010d 0f 00000000800012e8
-0 00800003ac 0007b783 0000010e 0f 0000000080002060
-0 00800003b0 0007b783 0000010f 0f 0000000000000040
-0 00800003b4 fef42623 00000110
-0 00800003b8 fec42783 00000111 0f 0000000000000040
-0 00800003bc 0007871b 00000112 0e 0000000000000040
-0 00800003c0 04000793 00000113 0f 0000000000000040
-0 00800003c4 00f70663 00000114
-0 00800003d0 00001797 00000115 0f 00000000800013d0
-0 00800003d4 f2078793 00000116 0f 00000000800012f0
-0 00800003d8 0007b783 00000117 0f 0000000080003060
-0 00800003dc 0007b783 00000118 0f 0000000000000040
-0 00800003e0 fef42623 00000119
-0 00800003e4 fec42783 0000011a 0f 0000000000000040
-0 00800003e8 0007871b 0000011b 0e 0000000000000040
-0 00800003ec 04000793 0000011c 0f 0000000000000040
-0 00800003f0 00f70663 0000011d
-0 00800003fc 00001797 0000011e 0f 00000000800013fc
-0 0080000400 efc78793 0000011f 0f 00000000800012f8
-0 0080000404 0007b783 00000120 0f 0000000080004060
-0 0080000408 0007b783 00000121 0f 0000000000000040
-0 008000040c fef42623 00000122
-0 0080000410 fec42783 00000123 0f 0000000000000040
-0 0080000414 0007871b 00000124 0e 0000000000000040
-0 0080000418 04000793 00000125 0f 0000000000000040
-0 008000041c 00f70663 00000126
-0 0080000428 00001797 00000127 0f 0000000080001428
-0 008000042c ed878793 00000128 0f 0000000080001300
-0 0080000430 0007b783 00000129 0f 0000000080005060
-0 0080000434 0007b783 0000012a 0f 0000000000000040
-0 0080000438 fef42623 0000012b
-0 008000043c fec42783 0000012c 0f 0000000000000040
-0 0080000440 0007871b 0000012d 0e 0000000000000040
-0 0080000444 04000793 0000012e 0f 0000000000000040
-0 0080000448 00f70663 0000012f
-0 0080000454 00001797 00000130 0f 0000000080001454
-0 0080000458 eb478793 00000131 0f 0000000080001308
-0 008000045c 0007b783 00000132 0f 0000000080006060
-0 0080000460 0007b783 00000133 0f 0000000000000040
-0 0080000464 fef42623 00000134
-0 0080000468 fec42783 00000135 0f 0000000000000040
-0 008000046c 0007871b 00000136 0e 0000000000000040
-0 0080000470 04000793 00000137 0f 0000000000000040
-0 0080000474 00f70663 00000138
-0 0080000480 00001797 00000139 0f 0000000080001480
-0 0080000484 e9078793 0000013a 0f 0000000080001310
-0 0080000488 0007b783 0000013b 0f 0000000080007060
-0 008000048c 0007b783 0000013c 0f 0000000000000040
-0 0080000490 fef42623 0000013d
-0 0080000494 fec42783 0000013e 0f 0000000000000040
-0 0080000498 0007871b 0000013f 0e 0000000000000040
-0 008000049c 04000793 00000140 0f 0000000000000040
-0 00800004a0 00f70663 00000141
-0 00800004ac 00001797 00000142 0f 00000000800014ac
-0 00800004b0 e6c78793 00000143 0f 0000000080001318
-0 00800004b4 0007b783 00000144 0f 0000000080008060
-0 00800004b8 0007b783 00000145 0f 0000000000000040
-0 00800004bc fef42623 00000146
-0 00800004c0 fec42783 00000147 0f 0000000000000040
-0 00800004c4 0007871b 00000148 0e 0000000000000040
-0 00800004c8 04000793 00000149 0f 0000000000000040
-0 00800004cc 00f70663 0000014a
-0 00800004d8 00001797 0000014b 0f 00000000800014d8
-0 00800004dc e4878793 0000014c 0f 0000000080001320
-0 00800004e0 0007b783 0000014d 0f 0000000080009060
-0 00800004e4 0007b783 0000014e 0f 0000000000000040
-0 00800004e8 fef42623 0000014f
-0 00800004ec fec42783 00000150 0f 0000000000000040
-0 00800004f0 0007871b 00000151 0e 0000000000000040
-0 00800004f4 04000793 00000152 0f 0000000000000040
-0 00800004f8 00f70663 00000153
-0 0080000504 00000513 00000154 0a 0000000000000000
-0 0080000508 0c8000ef 00000155 01 000000008000050c
-0 00800005d0 fd010113 00000156 02 00000000800eefa0
-0 00800005d4 02813423 00000157
-0 00800005d8 03010413 00000158 08 00000000800eefd0
-0 00800005dc 00050793 00000159 0f 0000000000000000
-0 00800005e0 fcf40fa3 0000015a
-0 00800005e4 f14027f3 0000015b 0f 0000000000000000
-0 00800005e8 fef43423 0000015c
-0 00800005ec fe843783 0000015d 0f 0000000000000000
-0 00800005f0 00379713 0000015e 0e 0000000000000000
-0 00800005f4 001027b7 0000015f 0f 0000000000102000
-0 00800005f8 00f707b3 00000160 0f 0000000000102000
-0 00800005fc fdf44703 00000161 0e 0000000000000000
-0 0080000600 00e78023 00000162
+0 00800002e8 00100513 000000e3 0a 0000000000000001
+0 00800002ec 2e4000ef 000000e4 01 00000000800002f0
+0 00800005d0 fd010113 000000e5 02 00000000800eefa0
+0 00800005d4 02813423 000000e6
+0 00800005d8 03010413 000000e7 08 00000000800eefd0
+0 00800005dc 00050793 000000e8 0f 0000000000000001
+0 00800005e0 fcf40fa3 000000e9
+0 00800005e4 f14027f3 000000ea 0f 0000000000000000
+0 00800005e8 fef43423 000000eb
+0 00800005ec fe843783 000000ec 0f 0000000000000000
+0 00800005f0 00379713 000000ed 0e 0000000000000000
+0 00800005f4 001027b7 000000ee 0f 0000000000102000
+0 00800005f8 00f707b3 000000ef 0f 0000000000102000
+0 00800005fc fdf44703 000000f0 0e 0000000000000001
+0 0080000600 00e78023 000000f1
```
</details>

WasabiFan, 2021-04-22:
From the first difference in the trace above, it's this assertion that's failing:

https://github.com/black-parrot-sdk/bp-tests/blob/ca7c052468e8f2cef93bc8ea86e525fbe710c1a6/src/uncached_mode_test.c#L61

dpetrisko, 2021-04-22:
Can you change this line: 
https://github.com/black-parrot-sdk/bp-tests/blob/master/src/uncached_mode_test.c#L7
to 0x80010000 instead?

My theory is that it's a coherence issue.  We're writing to program memory (DRAM base address) without a fence.i, which is odd. The difference in cache size would mean that the eviction pattern is different.  Hard to say if it's a test bug or a hardware bug without looking more at it. @muwyse ?

Also tagging @Shashank-Vijay  since he wrote the test. Maybe I'm misunderstanding it

muwyse, 2021-04-23:
I'll need to take another in-depth look at the test, but it looks like the program is switching the D$ back to uncached only mode during execution. I'm assuming that the NBF loader puts it into cached mode during startup because that would be the normal behavior. I'm not sure we ever really verified this could be done. It very well could be a coherence issue.

I think code usually gets placed at the DRAM base address, so it seems like this line (https://github.com/black-parrot-sdk/bp-tests/blob/ca7c052468e8f2cef93bc8ea86e525fbe710c1a6/src/uncached_mode_test.c#L24) would be clobbering an instruction?

We should definitely track down the underlying issue here. But, I also think the test structure should be changed to just declare some array and read and write from that array. Let the compiler choose the addresses we manipulate. If we put the D$ into uncached mode it will make uncached accesses to those addresses.

dpetrisko, 2021-04-23:
I thought about this a bit more: there should be a fence after each mode switch. We should still dig deeper into the bug, but that's a requirement since there's indeterminate behavior otherwise.

https://github.com/black-parrot-sdk/bp-tests/blob/ca7c052468e8f2cef93bc8ea86e525fbe710c1a6/src/uncached_mode_test.c#L24