black-parrot/black-parrot #849: bp_hprint: what is this function designed to do?

WasabiFan, 2021-04-23:
`perch` defines [`bp_hprint`](https://github.com/black-parrot-sdk/perch/blob/master/bp_utils.c#L48), which _seems_ to be intended to print a hex nibble. I assume this because it takes a value `hex` and prints the character computed as:

```c
*(PUTCHAR_BASE_ADDR) = ('0' + hex);
```

However, ASCII letters don't start immediately after the numbers -- there's some punctuation in the middle. So... what is this function intended to be used for? There are few callers in the black-parrot hierarchy, and it may indeed be that it's never been observed with values larger than 9.

dpetrisko, 2021-04-23:
It’s intended to be used to print a raw byte value. No one here is going to defend that it’s useful

WasabiFan, 2021-04-23:
Ahhh OK, so it's just for printing discernable values, without a particular format. Got it. Thanks!

For others' convenience, I've been using the following for printing larger values:

```c
void print_hex_nibble(uint8_t nibble) {
      if (nibble < 0xA) {
            bp_cprint('0' + nibble);
      } else {
            bp_cprint('a' + nibble - 0xa);
      }
}

void print_uint64(uint64_t val) {
      bp_cprint('0');
      bp_cprint('x');
      for (int i = 0; i < 16; i++) {
            uint8_t nibble = val >> 60;
            print_hex_nibble(nibble);
            val <<= 4;
      }
      bp_cprint('\n');
}
```

dpetrisko, 2021-04-23:
Sound like good additions to the library!

WasabiFan, 2021-04-23:
In that case, I'll leave this issue open for the time being -- I can probably open a PR with the above to perhaps replace the `bp_hprint` function. We'll see when I have some time.

closed 2021-04-30 by dpetrisko