black-parrot/black-parrot #854: Test higher DRAM space

dpetrisko, 2021-04-29:
Currently, our simulation environment tests with 32 bits of DRAM space. We should test with more. This is a little tricky within the context of VCS, but should be reasonable as a one-off test

cknizek, 2025-10-07:
Is this still relevant? I'd like to tackle it but only have Verilator access, not VCS. Would that work?

dpetrisko, 2025-10-07:
Hi, should be fine. The issue with VCS is that you can't instantiate a dense memory of 2**32 bits. Not sure if Verilator has this limitation as well. Regardless, using an sparse/associate/C++-based memory is probably a better approach anyway.

Could you do a quick survey of other open-source projects and see how they handle their backing DRAM? If there's anything generally accepted, we should use that.

So this issue basically has the two prongs:
- enable >32b memory simulations
- test >32b caddr_width configurations

Keep me posted!

dpetrisko, 2026-09-08:
I believe this is resolved with @HUNT-001 's PRs. Thank you!

PR #1145 (2022-11-30): Consolidating aviary configurations
## Summary
This PR cleans up the configurations in the aviary

## Issue Fixed
#854 

## Area
Aviary

## Reasoning (outdated, confusing, verbose, etc.)
Some users have requested the ability to scale to 64kB caches. PPA analysis should be done before taping out such a large cache

## Additional Changes Required (if any)
None

## Analysis
The configuration itself will not affect anything, but consolidating up the configs will allow us to better verify changes quicker

## Verification
Standard regression and modifying the regression to test the new configs

## A [... cut]
closed 2026-09-08 by dpetrisko