black-parrot/black-parrot #869: Remove internal message header buffering in CCE's (lce_req, lce_resp, mem_resp)

muwyse, 2021-05-18:
- [ ] Remove LCE Request header buffer
- [ ] Remove LCE Response header buffer
- [ ] Remove Memory Response header buffer (or determine if unable to)

muwyse, 2021-08-12:
Memory Response requires buffering to avoid deadlock in the coherence protocol. The CCE needs space for two messages on this channel since it may issue two memory commands prior to resolving the speculative issue of a cache block fetch.

dpetrisko, 2021-08-12:
Where should we document this?

muwyse, 2021-08-12:
I think best places are:
- https://github.com/black-parrot/black-parrot/blob/me_dev/docs/bedrock_coherence_protocol.md
- coherence deep dive doc (added by ME interface PR)
- comments in the code, at the stream pumps (I did add brief comments here, although they could be reworded to more explicitly state that the buffer size must be two elements)

dpetrisko, 2021-08-12:
Yeah the first one can maybe mention in that Priority section about how LCE cmd depends on mem cmd / mem resp being completed. 

I haven’t looked through the second one. 

The third, do you mean commenting at the CCE instantiation of stream pumps? I think that’s the correct place for saying exactly the sizing of the buffers since it’s implementation specific (you could theoretically have a different buffering strategy etc)

dpetrisko, 2023-07-20:
This is done

closed 2023-07-20 by dpetrisko