black-parrot/black-parrot #896: Swap out hardened RAM generics in bsg_sv2v flow

dpetrisko, 2021-07-14:
## Branch / Commit Hash
any

## Area
bsg_sv2v

## Current behavior
bsg_sv2v currently pickles the entire BlackParrot core without replacing the RAMs. This takes a decent amount of time ~20-30m. 

## Desired behavior
We should swap the RAMs with hardened generics. Additionally, this is a good excuse to have a formal list of the RAMs intended to be hardened.

## Proposed solution
- [ ] Add a list of hardened RAMs (in JSON format?)
- [ ] Add hardened generic wrappers / wrapper macros (may also live in basejump
- [ ] Modify bsg_sv2v flow to do hardened swap

## Analysis
Reduced runtime in bsg_sv2v.

## Verification
Manual inspection and runtime analysis

## Additional context
 [Example of hard wrapper](https://github.com/bespoke-silicon-group/bsg_pipeclean_suite/blob/master/black_parrot_fe_only_2019_03_11/harden_me.v))

taylor-bsg, 2021-07-17:
@stdavids i thought we had some kind of method for these hard rams? Was it part of IDF or something?

dpetrisko, 2021-07-17:
The methodology is to use this swap:

https://github.com/bespoke-silicon-group/basejump_stl/blob/master/hard/pickle_40/bsg_mem/bsg_mem_1rw_sync.v

and then include this hardened file for sims.

https://github.com/bespoke-silicon-group/bsg_pipeclean_suite/blob/master/black_parrot_fe_only_2019_03_11/harden_me.v

For the actual sv2v here, we just have to swap out the memory and the wrapper will be blackboxed. If we want to sim the sv2v, then we need the wrapper functional modules to simulate with.

PR #903 (2021-07-21): Adding hardened stubs
## Issue Fixed
#896 

## Area
Makefile.sv2v, adds common verilog files

## Reasoning (outdated, confusing, verbose, etc.)
The current SV2V flow does not substitute blackbox RAMs, leading to long runtimes and excessively long elaboration lists, causing issues in some tools.

## Additional Changes Required (if any)
none

## Analysis
No impact on PPA, some makefile complexity. 20m->1m bsg_sv2v runtime.

## Verification
Manual inspection of output netlist.

## Additional Context
none
closed 2021-07-21 by dpetrisko