black-parrot/black-parrot #959: L1 atomics broken in multicore configs

muwyse, 2021-11-05:
## Summary
L1 atomics appear to be broken (according to mc_* tests from bp-tests) in the multicore configurations. Needs investigation. L1 atomics will be important for decent performance when booting SMP Linux.

## Branch / Commit Hash
https://github.com/black-parrot/black-parrot/tree/me_dev_fix_l1amo

## Verification
ci/multicore.sh should pass with L1 atomics enabled for multicore configurations.

dpetrisko, 2021-11-05:
What test is failing on what config?

dpetrisko, 2021-11-05:
And do L2 atomics work?

dpetrisko, 2021-11-06:
Oh and last question: Is it a cosim error or a true error?

muwyse, 2021-11-11:
> Is it a cosim error or a true error?
Unsure. There was a config issue where amo_en_p wasn't getting set properly (thus Dromajo was running without --enable_amo) but that should be fixed now. Based on quick glance at the output most failures appear to be mismatch of write value returned by an amoswap.d instruction while attempting to acquire a global lock. Perhaps this is due to a re-ordering of amo operations being executed by the various cores and some order mismatch between Dromajo and RTL sim?

> Do L2 atomics work?
Have not checked, but I don't believe the LCE and CCE have this functionality implemented yet.

> Which test is failing on which config?
all three of the mc_* tests across various core counts and both CCE configs. I'm waiting for multicore CI to run again so I can debug.

dpetrisko, 2022-01-02:
Fixed with #993

closed 2022-01-02 by dpetrisko