black-parrot/black-parrot #961: simsc segfault

jackhumphries, 2021-11-14:
## Branch / Commit Hash
What version of BlackParrot are you running?

a4779726d0cc5d3427d2d26cb5ae8665c9361375

I am running this on Ubuntu 20.04.

## Area
Module/tool/makefile/script in question.

simsc command when running the tire kick:

```
cd /home/jhumphri/black-parrot-sim/rtl/bp_top/syn/results/verilator/bp_tethered.e_bp_default_cfg.none.sim.bp-tests.hello_world; ./simsc   2>&1 | tee /home/jhumphri/black-parrot-sim/rtl/bp_top/syn/logs/verilator/bp_tethered.e_bp_default_cfg.none.sim.bp-tests.hello_world.log
```

## Current behavior
A clear and concise description of what is happening.
Steps to reproduce the behavior:

simsc segfaults before printing the CORE0 stats:

```
...
NBF loader done!
Hello world!
[CORE0 FSH] PASS
All cores finished! Terminating...
- /home/jhumphri/black-parrot-sim/rtl/bp_top/test/common/bp_nonsynth_host.sv:221: Verilog $finish
Finishing test
Executing final
Segmentation fault
[CORE0 FSH] PASS
make: [/home/jhumphri/black-parrot-sim/rtl/bp_top/test/tb/bp_tethered/Makefile.verilator:119: /home/jhumphri/black-parrot-sim/rtl/bp_top/syn/results/verilator/bp_tethered.e_bp_default_cfg.none.sim.bp-tests.hello_world/tire_kick] Error 1 (ignored)
make: Leaving directory '/home/jhumphri/black-parrot-sim/rtl/bp_top/syn'
```

`dmesg` clearly indicates that the segfault is in simsc:
```
[82464.603081] simsc[296099]: segfault at 0 ip 00007fa702a98f5b sp 00007ffc55b0c9d0 error 4 in libc-2.31.so[7fa702a39000+178000]
[82464.603091] Code: 08 00 64 44 89 23 85 c0 75 d4 e9 2b ff ff ff e8 9b bd 00 00 e9 fd fe ff ff e8 b1 db 0a 00 90 f3 0f 1e fa 41 54 55 48 89 fd 53 <8b> 07 f6 c4 20 0f 85 ea 00 00 00 89 c2 81 e2 00 80 00 00 0f 84 e9
```

I noticed that removing the output redirection to `tee` prevents the segfault and lets simsc print the results properly:
```
./simsc   2>&1
```

When I run simsc under GDB with the redirection of output to `tee`, I see this crash trace:

```
NBF loader done!
Hello world!
[CORE0 FSH] PASS
All cores finished! Terminating...
- /home/jhumphri/black-parrot-sim/rtl/bp_top/test/common/bp_nonsynth_host.sv:221: Verilog $finish
Finishing test
Executing final

Thread 1 "simsc" received signal SIGSEGV, Segmentation fault.
0x00007ffff78d8f5b in _IO_new_fclose (fp=0x0) at iofclose.c:48
48	iofclose.c: No such file or directory.
(gdb) bt
#0  0x00007ffff78d8f5b in _IO_new_fclose (fp=0x0) at iofclose.c:48
#1  0x00005555555672e3 in VerilatedContextImp::fdClose(unsigned int) ()
#2  0x00005555556fc6af in Vtestbench___024root___final_TOP(Vtestbench___024root*) ()
#3  0x00005555556a520f in Vtestbench___024root___final(Vtestbench___024root*) ()
#4  0x000055555555afe6 in main ()
```

## Expected behavior
A clear and concise description of what you expected to happen.

The test should finish and print the results as expected.

## Proposed solution
Potential solution(s) including scope of work and affected modules.

I haven't dug through this deeply. It's unclear to me whether this is definitely an issue with BlackParrot or maybe instead an issue with Verilator.

## Analysis
Potential impact of the bugfix.

Makes setup easier.

## Verification
Steps to verify the bugfix.

```
cd /home/jhumphri/black-parrot-sim/rtl/bp_top/syn/results/verilator/bp_tethered.e_bp_default_cfg.none.sim.bp-tests.hello_world; ./simsc   2>&1 | tee /home/jhumphri/black-parrot-sim/rtl/bp_top/syn/logs/verilator/bp_tethered.e_bp_default_cfg.none.sim.bp-tests.hello_world.log
```

The above command from tire kick should run and print the results without crashing.

## Additional Context
Add any other context about the problem here.
Logs, screenshots or videos that show the issue are very helpful.

black-parrot, 2021-11-15:
> to @black-parrot <https://github.com/black-parrot>.

Maybe share your verilator version #

dpetrisko, 2021-11-15:
Based on https://github.com/black-parrot/black-parrot-sim/issues, it seems like you're using the apt version of Verilator, which is quite old. The instructions in the readme here: https://github.com/black-parrot/black-parrot-sim have you run 'prep_lite' which will build a more recent version of verilator as part of the process. Can you try that and see if the issue persists?

jackhumphries, 2021-11-15:
I grabbed the latest version of verilator and the issue still occurs.

```
# verilator --version
Verilator 4.214 2021-10-17 rev v4.214-21-g0185ee5d
```

dpetrisko, 2021-11-26:
Looks to be an Ubuntu issue, can't reproduce this on Centos

dpetrisko, 2023-07-18:
Closing for staleness, reopen if still occurring on Verilator 5+

closed 2023-07-18 by dpetrisko