black-parrot/black-parrot #967: BP does not raise exception when attempting to update mhartid

scanakci, 2021-11-25:
## Branch / Commit Hash
What version of BlackParrot are you running?
Commit: 5d6620f4bf5f6a31b3ecbb3dfdf92ef485a9aa6c

## Area
BlackParrot RTL

## Current behavior and Expected behavior
When attempting to update mhartid register with csrrci instruction ( csrrci  a1,mhartid,14 ), BP should raise an exception according to spec since mhartid is a read-only register. However, it does not. Dromajo, spike both raise exception. I attached a folder that includes the necessary elf file along with all dumps generated by BP during simulation. 

## Proposed solution
N/A

## Analysis
N/A

## Verification
1. Copy prog.riscv into black-parrot-sim/rtl/../sdk/prog/bp-tests/ 
2. Run "make -B build.sc sim.sc COSIM_P=1 CMT_TRACE_P=1 SUITE=bp-tests PROG=prog" in black-parrot-sim/rtl/bp_top/syn

## Additional Context
Dromajo output:
Running with Dromajo cosimulation
[error] HARTID: 0
[error] EMU PC 0000000080000144, DUT PC 00000000800002e0
[error] EMU INSN 00001097, DUT INSN f14775f3
[error] EMU WDATA 0000000080001144, DUT WDATA 0000000000000000
[error] EMU MSTATUS a00005800, DUT MSTATUS a00004800
[error] DUT pending exception -1 pending interrupt -1
NBF loader done!
COSIM_FAIL: instruction mismatch
[mhartid_mismatch.tar.gz](https://github.com/black-parrot/black-parrot/files/7599732/mhartid_mismatch.tar.gz)

dpetrisko, 2021-11-25:
Relevant thread: https://groups.google.com/a/groups.riscv.org/g/isa-dev/c/gGRGSL2vOIo/m/R8SnwnpqBAAJ?utm_medium=email&utm_source=footer

This is probably true for other read-only CSRs which were presumed to be WARL

dpetrisko, 2022-01-02:
Fixed with #968

PR #968 (2021-12-24): Various CSR fixes
This is a patch to fix: #967, #969, #970

Readonly CSRs must trap. That includes mhartid: https://groups.google.com/a/groups.riscv.org/g/isa-dev/c/gGRGSL2vOIo/m/R8SnwnpqBAAJ?utm_medium=email&utm_source=footer

There's also a fix so that FS is not dirtied during an illegal instruction trap.

There's also a patch to hardcode the low 2 bits of MEPC and SEPC

Tests for all in an incoming PR
closed 2022-01-02 by dpetrisko