black-parrot/black-parrot #971: Some FPU operations not producing NaNs as they should

mysoreanoop, 2021-11-28:
## Branch / Commit Hash
a4779726d0cc5d3427d2d26cb5ae8665c9361375
## Area
FPU submodules
## Current behavior
Some operations (part of a contrived test assembly code) which would evaluate to NaN in Spike/Dromajo/Rocket are not being evaluated to NaN in BP.
(BP doesn't seem to infer NaNs from SP values read off of DP registers. As a result, any operation which should have otherwise seen NaN as one of the operands will operate normally producing a non-NaN value.)

## Expected behavior
Conformance with Spike/Dromajo.
(When running SP operations from SP values inferred off of DP registers, if the SP values are not explicitly boxed apriori, the operation has to infer the SP operand values as NaNs (barring the coincidental boxed value maybe?). Here's some relevant discussion: https://groups.google.com/a/groups.riscv.org/g/isa-dev/c/y8CkRopfEWk)

## Proposed solution
Check if operands are boxed or not, and if not, consider them to be NaNs (if not already doing so)

## Verification
Running the same test case against Dromajo/Spike should match the commit traces.

## Additional Context
The table below reads the rd for Black Parrot, 
Dromajo, Spike, Rocket, and rs1, rs2, rs3 for 2 operations (fnmadd, and fdiv -- there are more similar mismatches for most FPU operations). Subsequent lines read the decoded value of the line above it
[image]
[fail.tar.gz](https://github.com/black-parrot/black-parrot/files/7612856/fail.tar.gz)

dpetrisko, 2022-01-05:
I believe this issue should be resolved with latest dev

black-parrot, 2022-01-05:
Do we have tests for these cases checked in?

dpetrisko, 2022-02-17:
Yes, tests commited with #1007

PR #1001 (2022-01-02): FPU refactor
## Summary
This PR refactors the FPU of blackparrot for better PPA and better compliance to IEEE.

## Issue Fixed
#971 

## Area
FPU, FP->REC, REC->FP

## Reasoning (outdated, confusing, verbose, etc.)
The BP FPU does all operations at 64b, and upconverts SP values to DP values upon load to the FP regfile. The previous FPU had a point of SP rounding after every operation such that we maintain correct fflags and precision. However, by storing the SP rounding mode in the regfile itself, it is possible to delay rounding until user-visible transfers (int regs or memory).

## Additiona [... cut]
PR #1007 (2022-01-10): Fixing nanboxing
## Summary
Fixes nanboxing behavior in the FMA and FDIV units. This is (technically?) a regression since #1001 brought in this faulty nanboxing code. However, the nanboxing box existed before this as well.

## Issue Fixed
#971

## Area
FPU

## Reasoning (outdated, confusing, verbose, etc.)
When doing a single precision operation, the operand will be converted into a NaN if the upper 32b of the raw value are not all 1. This is called nan-boxing in RISC-V terminology. We did attempt to nan-box, but we injected the incorrect NaN when unboxing. This caused us to accidentally garba [... cut]
closed 2022-02-17 by dpetrisko