black-parrot/black-parrot #973: paging.c from bp-tests not working

jackhumphries, 2021-12-01:
## Branch / Commit Hash
What version of BlackParrot are you running?

I am running commit 7d53958b950a122e50345e1f45c07ddcc6ecd1db.

## Area
Module/tool/makefile/script in question.

bp-tests/src/paging.c

## Current behavior
A clear and concise description of what is happening.
Steps to reproduce the behavior:

When I run paging.c (`make -C bp_top/syn build_dump.sc sim_dump.sc SUITE=bp-tests PROG=paging CFG=e_bp_multicore_1_cfg`), I get the following crash:

```
NBF loader done!                                                                                                      
trap_entry kva: 0xffffffffffe00150                                                                                    
mcause: 0x0000000000000002                                                                                            
mtval:  0x0000000000000000                                                                                            
mepc:   0x0000000000000540                                                                                            
Illegal instruction                                                                                                   
[CORE0 FSH] FAIL                                                                                                      
All cores finished! Terminating...                                                                                    
- /home/jhumphri/black-parrot-sim/rtl/bp_top/test/common/bp_nonsynth_host.sv:221: Verilog $finish                     
Finishing test                                                                                                        
Executing final                                                                                                       
[CORE0 STATS]                                                                                                         
        clk   :                11873                                                                                  
        instr :                 5488                                                                                  
        mIPC  :                  462
Closing dump file
Exiting
```

0x540 is the physical address for userTest():
```
0000000080000540 <userTest>:                                                          
    80000540: fe010113            addi  sp,sp,-32                                     
    80000544: 00813c23            sd  s0,24(sp)                                       
    80000548: 02010413            addi  s0,sp,32                                      
    8000054c: 00100793            li  a5,1                                            
    80000550: fef42623            sw  a5,-20(s0)                                      
    80000554: 0280006f            j 8000057c <userTest+0x3c>                          
    80000558: fec42703            lw  a4,-20(s0)                                      
```

## Expected behavior
A clear and concise description of what you expected to happen.

The test should complete successfully.

## Proposed solution
Potential solution(s) including scope of work and affected modules.

I am not sure what the issue is.

## Analysis
Potential impact of the bugfix.

The test is not working.

## Verification
Steps to verify the bugfix.

See above.

## Additional Context
Add any other context about the problem here.
Logs, screenshots or videos that show the issue are very helpful.

dpetrisko, 2021-12-01:
Hi, thanks for the issue. I would suggest adding COSIM_P=1 to get a clearer picture of what’s going wrong

jackhumphries, 2021-12-01:
Hi Dan,

Thanks for the quick response. When I run with `COSIM_P=1`, I now see this output:
```NBF loader done!
[DEBUG] DUT raised exception 12
[DEBUG] DUT raised exception 2
mcause: 0x0000000000000002
mtval:  0x0000000000000000
mepc:   0x0000000000000540
Illegal instruction
[CORE0 FSH] FAIL
All cores finished! Terminating...
```
Exception 12 is an instruction page fault, so I suppose it faults upon accessing the first instruction of `userTest()`?

A few (stupid) questions:
1. `tf.epc = (uint64_t)userTest - DRAM_BASE;` is set to the physical address of `userTest()`. Does this not need to be a virtual address?
2. Why do we need to `pop_tf()` on the trapframe before running `userTest()`?

I am still parsing through the test, but please let me know if you have a sense of the issue.

dpetrisko, 2021-12-01:
Hi, I've identified the problem.  paging.c writes to mega/gigapages as you've noted. The default configuration of BP does not have gigapage support (itlb_els_1g: 0) in the aviary. Now, this should be totally fine and the implementation should store mega/gigapages into 4k entries. 

However, there is a bug where if you write a gigapage without gigapage support, the hardware will not fill it in the TLB at all. We will fix this, but in the meantime I would suggest applying this patch:

```
--- a/bp_common/src/include/bp_common_aviary_pkgdef.svh
+++ b/bp_common/src/include/bp_common_aviary_pkgdef.svh
@@ -238,8 +238,8 @@

       ,itlb_els_4k : 8
       ,dtlb_els_4k : 8
-      ,itlb_els_1g : 0
-      ,dtlb_els_1g : 0
+      ,itlb_els_1g : 1
+      ,dtlb_els_1g : 1

       ,lr_sc                : e_l1
       ,amo_swap             : e_none
```

Thanks for the find!

dpetrisko, 2021-12-01:
@farzamgl may be able to explain better since he wrote it, but 

>tf.epc = (uint64_t)userTest - DRAM_BASE; is set to the physical address of userTest(). Does this not need to be a virtual address?

I believe the mapping is simply vaddr = paddr - 0x80000000 and so that is a virtual address

jackhumphries, 2021-12-01:
Dan, you rock! Thanks for responding this late at night. I am running this now and will let you know if it works on my end, too.

To further support what you said, I noticed that `mapping.c` works up  until it starts writing to the 1G page at the halfway point of the first for loop in `userTest()`. In other words, when I run with Dromajo cosimulation, I see exception #15 outputted PTES_PER_PT / 2 = 256 times, which is expected since the mapping is changed to a 1G page at the halfway point of the loop.

jackhumphries, 2021-12-01:
Hi Dan,

I can confirm that the test passes now. Thanks again!

I'm still parsing the test, but I understand now that `(uint64_t)userTest - DRAM_BASE` is a virtual address. And when we get an exception 12 (instruction fetch page fault) on the first instruction in `userTest()`, the fault is handled ultimately by `gigapage_init()` since in virtual address 0x540, `vpn2(addr) % 3` is 0. And without the patch, the test was crashing on that instruction fetch due to the gigapage issue.

I will wait for @farzamgl to explain the trapframe stuff. After more reading, I se that `pop_tf()` issues an `sret` instruction, which puts the CPU into user mode since `SPP` is set to 0 in the `mstatus` register. Is that the point of this?

I notice the following line in `main()`:
`l1pt[PTES_PER_PT-1] = ((uint64_t)kpt >> PGSHIFT << PTE_PPN_SHIFT) | PTE_V;`

I am confused as to why this in not:
`l1pt[PTES_PER_PT-1] = ((uint64_t)kva2pa(kpt) >> PGSHIFT << PTE_PPN_SHIFT) | PTE_V;`

The second one would be in line with how page tables are filled in by `page_init()`, `megapage_init()`, etc.

Thanks for your help!

farzamgl, 2021-12-01:
Hi Jack,

> 1. `tf.epc = (uint64_t)userTest - DRAM_BASE;` is set to the physical address of `userTest()`. Does this not need to be a virtual address?

That is the virtual address. For this simple VM test, we've mapped the user mode vaddr to "paddr - DRAM_BASE" and the kernel vaddr to "paddr - DRAM_BASE - MEGAPAGE_SIZE". So the 0x0 address in user mode is mapped to 0x80000000.

> 2. Why do we need to `pop_tf()` on the trapframe before running `userTest()`?

The reason is that we're starting the main function in machine mode and we need to jump to the user mode with an `eret` instruction so we use pop_tf() which is mainly used to return from traps to the machine mode and we just manually set the trap frame before calling it in the main.

> I notice the following line in main():
> l1pt[PTES_PER_PT-1] = ((uint64_t)kpt >> PGSHIFT << PTE_PPN_SHIFT) | PTE_V;
> 
> I am confused as to why this is not:
> l1pt[PTES_PER_PT-1] = ((uint64_t)kva2pa(kpt) >> PGSHIFT << PTE_PPN_SHIFT) | PTE_V;

That's because we're in the machine mode and the pointer to the page of the page-table that contains the kernel translations(kpt) contains a physical address value, so there's no need to use kva2pa().

jackhumphries, 2021-12-02:
Thanks for your response Farzam. I read your response and have been staring at the code, but I am still a bit confused as to the third question (and I haven't worked on a kernel memory subsystem since undergrad :) ). Part of the issue is that I can't print anything out once the CPU enters user mode since I am unable to set up the mappings for the addresses in `host.h`.

When the user mode triggers a page fault, the fault is handled by `handle_trap()`, which runs in kernel mode. The kernel mode is presumably also using virtual addressing.

In `page_init()`, consider this line:
```
l1pt[vpn2(vaddr)] = (((uint64_t)kva2pa(l2pt) >> PGSHIFT) << PTE_PPN_SHIFT) | PTE_V;
```

If the kernel is using virtual addressing mode, then why can it directly dereference `l1pt`. Isn't `l1pt` a pointer with a physical address? Furthermore, why do we have `kva2pa(l2pt)` instead of just `l2pt`? Similarly, isn't `l2pt` also a physical address already?

Thanks for your help!

farzamgl, 2021-12-02:
Good question!
The reason that the l1pt symbol is translated to its kernel address when we're running kernel code in page_init() is because of the `-mcmodel=medany` flag used for compilation. With `medany` any reference to global variables is done through PC relative addressing(`auipc`/`ld` pairs) instead of absolute addressing, and since PC has a virtual address in kernel mode, we get the virtual address of the page-table symbols.
I understand that this is not a very clean way of doing this. Usually privileged code and machine code are kept in different files and page-table addresses are generated independently and not from a global variable.

jackhumphries, 2021-12-02:
Thanks Farzam! This makes sense. I have a couple more questions, if you have a moment. Thanks again -- I appreciate it.

1. Writing to `satp` turns virtual addressing on, but I assume that the CPU continues to use physical addressing while it is in machine mode. So `pop_tf()`, which moves us into user mode, also triggers the CPU to start using virtual addressing. Is this correct?

2. How are the PTEs created for the memory that backs the binary instructions? I notice that exception 12 (instruction page fetch) is triggered as soon as the CPU enters user mode and tries to execute `userTest()`, but why does the kernel not in turn trigger a page fault (i.e., exception 12)? I see that you map `trap_entry` to a kernel virtual address and then write the kernel virtual address in the `stvec` CSR (so it makes sense that fetching `trap_entry()` instructions does not trigger a page fault), but `trap_entry` has the instruction `j 80000580 <handle_trap>` in the disassembled output. Shouldn't the CPU experience a page fault on the first instruction of `handle_trap()`?

farzamgl, 2021-12-02:
> 1. Writing to `satp` turns virtual addressing on, but I assume that the CPU continues to use physical addressing while it is in machine mode. So `pop_tf()`, which moves us into user mode, also triggers the CPU to start using virtual addressing. Is this correct?

Yes.

> 2. How are the PTEs created for the memory that backs the binary instructions? I notice that exception 12 (instruction page fetch) is triggered as soon as the CPU enters user mode and tries to execute `userTest()`, but why does the kernel not in turn trigger a page fault (i.e., exception 12) when it starts running its own instructions? I see that you map `trap_entry` to a kernel virtual address and then write the kernel virtual address in the `stvec` CSR (so it makes sense that fetching `trap_entry()` instructions does not trigger a page fault), but `trap_entry` has the instruction `j 80000580 <handle_trap>` in the disassembled output. Shouldn't the CPU experience a page fault on the first instruction of `handle_trap()`? Or does it not because the `j` instruction is PC-relative?

Yes. The jump instruction is also translated to a `jal x0, offset` that's PC relative, so it'll jump to the kernel virtual address. It doesn't throw a page-fault because we've already mapped the kernel megapage in the main function. That way the entire code is mapped to the first 2MB of the memory and we can run the trap handler without any page-faults.

jackhumphries, 2021-12-02:
Thanks Farzam! That's all the questions I have for now.

I'm happy to close this issue, but not sure if Dan wants to keep it open until the fix is applied for the 1G pages.

dpetrisko, 2021-12-02:
Yeah leave it open, I’ll close when the PR is merged. Thanks!

dpetrisko, 2022-01-02:
Closing, fixed with #975

PR #975 (2021-12-24): Fixing behavior for zero gigapage support
## Summary
This PR makes the TLB fall back to kilopages when gigapage support is not available. The old behavior was to drop the gigapage which is obviously not correct.

## Issue Fixed
#973 

## Area
TLB

## Analysis
Should be no PPA impact, since the expression is not critical

## Verification
Ran paging test before and after the change with and without gigapage support.  Adding this test to the regression before merging
closed 2022-01-02 by dpetrisko