black-parrot/black-parrot #995: BP ignores misaligned memory access exception in S-mode

Yuan-Mao, 2021-12-28:
## Branch / Commit Hash

Current master branch(dd2d8453a49b0f069e96f175afe97968dac205e1)

## Area
Misaligned Memory Access

## Current & Expected behavior
Hi,
When I was trying to bring up Linux kernel 5.15 on BP, I found a mismatch between Dromajo and BP during co-simulation. It happens when BP does its first misaligned memory access in S-mode. Dromajo will jump to M-mode and handle the memory access exception properly, but BP seems to just ignore it and continue its execution.

[This code from Linux 5.15](https://elixir.bootlin.com/linux/v5.15/source/arch/riscv/lib/memmove.S#L6) is the code that reveals this issue. Back in kernel 5.7, there's no such assembly code so every memory copy is memory aligned, but in kernel 5.15 we can easily have plenty of unaligned memory access with this piece of code (it uses lw, sw even if src and dst are not 4-byte aligned).

To reproduce this issue, I have written a program that does misaligned memory access. Please find the attached files 'opensbi.elf' and 'opensbi.nbf' here: [test.zip](https://github.com/black-parrot/black-parrot/files/7782667/test.zip).
'opensbi.elf' is basically an opensbi image with a simple payload(prog.c) that uses the memmove function(memmove.S) from [Linux 5.15](https://elixir.bootlin.com/linux/v5.15/source/arch/riscv/lib/memmove.S#L6) to do some misaligned memory access.

[image]

dpetrisko, 2021-12-28:
Hi, that makes sense we don't currently detect misaligned exceptions in any priv mode as these are exceptionally rare code. Interesting that Linux now does this. Please make a PR to [bp-tests](https://github.com/black-parrot-sdk/bp-tests) with the failing minimal test case

dpetrisko, 2021-12-29:
Hi, I have added support here: 
https://github.com/black-parrot/black-parrot/pull/997
https://github.com/black-parrot-sdk/bp-tests/pull/26

dpetrisko, 2022-01-05:
Fixed

PR #997 (2022-01-01): Misaligned Memory Access Detection

## Summary
This PR adds misaligned instruction / load / store detection to BP. Upon detection the processor will trap as a normal exception.

## Issue Fixed
#995 

## Area
bp_mmu, bp_be_pipe_mem, bp_fe_top

## Reasoning (outdated, confusing, verbose, etc.)
Trapping on misaligned instructions and load/stores is required for RISC-V compliance.

## Additional Changes Required (if any)

## Analysis
Some minor area overhead for the size decoding logic, but this is necessary for RISC-V compliance. Although performant programs should not use this, some low level co [... cut]
closed 2022-01-05 by dpetrisko