black-parrot/black-parrot #996: Support TM bit in SCOUNTEREN

Yuan-Mao, 2021-12-28:
## Branch / Commit Hash
What version of BlackParrot are you running?

dd2d8453a49b0f069e96f175afe97968dac205e1

## Area

Support TM bit in SCOUNTEREN

## Current behavior

Linux kernel 5.15 seems to use rdtime in U-mode in its vDSO implementation, so before going into U-mode we need the TM bit set in SCOUNTEREN, but it seems to me that currently we do not support this both in BP and Dromajo.

dpetrisko, 2021-12-28:
mtime is a memory-mapped CSR, so accesses to it at any level must trap to M-mode

Yuan-Mao, 2021-12-28:
It seems to me that rdtime does not access mtime. It uses 'time' CSR instead, and that CSR read needs the TM bit set if we don't want to get any illegal instruction exception.
[image]
[image]

dpetrisko, 2021-12-29:
As that clip says, the time csr is a read-only shadow of mtime. mtime cannot be read with a CSR instruction, it can only be read by MMIO, regardless of privilege level. Trapping and letting opensbi handle the access is how all cores that I know of handle this

Yuan-Mao, 2021-12-29:
I see. Then we have to count on opensbi to handle that. [The opensbi version](https://github.com/black-parrot-sdk/opensbi/tree/eac24d8778eb8497629431af523d05fd54b42d42) now using in our SDK does not handle the time CSR access emulation properly, but the latest version seems to be able to handle that. Maybe we will want to update our opensbi to newer version.

dpetrisko, 2022-01-02:
Closing this, tracking here: https://github.com/black-parrot-sdk/black-parrot-sdk/issues/30

closed 2022-01-02 by dpetrisko