black-parrot/black-parrot #1002: CLINT Memory Map in BP

Yuan-Mao, 2022-01-02:
## Branch / Commit Hash
I'm running dd2d8453a49b0f069e96f175afe97968dac205e1

## Area
ACLINT memory map V.S. Current CLINT memory map in BP

## Background information gathered
Hi,
From what I have learned from the [riscv aclint spec](https://github.com/riscv/riscv-aclint/blob/main/riscv-aclint.adoc#23-mtimecmp-registers-offsets-0x00000000---0x00007ff0), we should have adjacent memory region for registers of the same kind. For example, mtimecmp from hart0 has offest 0x0 while mtimecmp from hart1 has offest 0x8. However, it seems to me in our current BP implementation, tile0 and tile1 would have 0x100_0000 offset difference between any kind of clint registers. 

## Question
Is our current implementation of BP conforming to the ACLINT spec? Or am I having any wrong idea? Thanks!

dpetrisko, 2022-01-02:
BP is not compliant to the aclint spec. Our fork of opensbi has a modified clint driver to support this

dpetrisko, 2022-01-02:
Do more recent versions of opensbi or linux use the SSWI or SETSSIP functionality of the ACLINT? Only base CLINT support used to be necessary, but things may have changed

Yuan-Mao, 2022-01-02:
Opensbi (v1.0) now conforms to ACLINT and we might need more modification than our current fork of opensbi to support it.
I haven't seen opensbi and linux use SSWI or SETSSIP. Opensbi only uses mswi and mtimer.

Yuan-Mao, 2022-01-02:
mswi: https://github.com/riscv-software-src/opensbi/blob/master/lib/utils/ipi/aclint_mswi.c
mtimer: https://github.com/riscv-software-src/opensbi/blob/master/lib/utils/timer/aclint_mtimer.c

dpetrisko, 2022-01-02:
Yeah we should be able to port our clint driver modifications to mswi and mtimer directly since those are backwards compatible with the CLINT: https://github.com/black-parrot-sdk/opensbi/commit/eac24d8778eb8497629431af523d05fd54b42d42

dpetrisko, 2022-01-02:
I'll leave this open as part of https://github.com/black-parrot/black-parrot/issues/989 since ACLINT support has been ratified