black-parrot/black-parrot #1009: [Feature Request] S-mode External Interrupt Line

Yuan-Mao, 2022-01-11:
## Branch / Commit Hash
dd2d845

## Area
S-mode external interrupt support

## Current behavior
Only M-mode external interrupt line is supported now (through the redundant memory-mapped plic bit)

## Desired behavior
Supporting S-mode external interrupt line can be useful because the [PLIC driver in Linux](https://elixir.bootlin.com/linux/v5.15/source/drivers/irqchip/irq-sifive-plic.c) leverages the S-mode interrupts(only when Linux is set to run in S-mode, which is usually the case).

My reasoning why this feature makes sense:
There are 3 possible ways to support PLIC driver in Linux:

1. The first one would be making PLIC send M-mode interrupt to Opensbi and then it sets the S-mode ext pending bit for Linux. This method does no work if we want to run opensbi and Linux kernel without any custom patches, because otherwise we would then need to somehow call into opensbi and clear the S-mode pending bit for Linux.
2. We can again make PLIC send M-mode interrupt to Opensbi, but this time delegate the M-mode ext interrupt to Linux. In theory this should work, but according to [this line](https://elixir.bootlin.com/linux/v5.15/source/drivers/irqchip/irq-sifive-plic.c#L321), when Linux parses the interrupt info in device tree, it would ignore any interrupts that are attached to M-mode external interrupt. Therefore if we want this method to work, we will then have to trick the kernel by writing an inconsistent device tree that shows S-mode interrupt in the software but actually it is an M-mode interrupt in the hardware.
3. The third one is to support S-mode ext interrupt, just the one I mentioned above.

## Proposed solution
I think for now we can do exactly the same thing for the S-mode interrupt: we can add another fake plic bit for S-mode, and change [this line](https://github.com/black-parrot/black-parrot/blob/dd2d8453a49b0f069e96f175afe97968dac205e1/bp_be/src/v/bp_be_calculator/bp_be_csr.sv#L142) to something like `wire sei_v = mie_r.seie & (mip_r.seip | external_s_mode_irq_i)`.

Relevant note that I find useful from the spec:

[image]

dpetrisko, 2022-01-11:
Adding the wire and a second external interrupt bit to the CLINT should be fine

dpetrisko, 2022-02-17:
Merged with #1028

closed 2022-02-17 by dpetrisko