black-parrot/black-parrot #1013: bp_me_axil_client module drops mesg

Yuan-Mao, 2022-01-16:
## Branch / Commit Hash
top_dev (a6a0d84)

## Area
bp_me_axil_client module:
https://github.com/black-parrot/black-parrot/blob/top_dev/bp_me/src/v/network/bp_me_axil_client.sv#L121

## Current behavior
If state_r == e_wait, and a master is sending an AXIL write to its client, then according to line 125~127, master should be able to get all the required ready signals to complete its write operation. However, if at this moment io_cmd_ready_and_i is not ready yet, then the data that meant to be sent will be lost.

dpetrisko, 2022-01-16:
Thanks for the report, aware of the issue. Should be fixed this weekend

dpetrisko, 2022-01-18:
See #1014

Yuan-Mao, 2022-01-18:
Thank you so much!

dpetrisko, 2022-02-17:
Fixed

closed 2022-02-17 by dpetrisko