black-parrot/black-parrot #1017: bp_me_axil_client use wstrb even if wvalid == 1'b0

Yuan-Mao, 2022-01-25:
## Branch / Commit Hash
860afac

## Area
bp_me_axil_client.sv

## Current behavior
https://github.com/black-parrot/black-parrot/blob/top_dev/bp_me/src/v/network/bp_me_axil_client.sv#L108

It seems to me the s_axil_wstrb_i is used even if s_axil_wvalid_i is not valid. If master issues a read and s_axil_wstrb_i happens to carry a valid value, then it might overwrite the read data width that should be fixed to the AXIL bus width.

dpetrisko, 2022-01-25:
Makes sense. And the read width is fixed to the bus width?

Yuan-Mao, 2022-01-25:
Yeah to my understanding the spec says the read width should be fixed to the bus width. Currently in our implementation, our read width is kind of depending on the wstrb value.

closed 2022-03-03 by dpetrisko